x-station/xstation-issues #262: Another "Please Insert Playstation CD-ROM" Issue

Chris32895, 2022-12-06:
OK so I feel like I've tried everything. I have resoldered all my points on the QSB. My PSX is a SCPH-5501 model. Leaving the lid closed with a formatted Sandisk Ultra Plus A1 32GB SD card gives me the "Please Insert Playstation CD-ROM" prompt. It blinks green twice beforehand. With no sd card it goes straight to the bios screen. I have tried Fat32 and exfat and multiple different firmware versions. I did notice one time that after booting and transferring my sd card back to my pc there was a config file in my 00xstation folder so there is that. I have checked continuity for all ground points and point 3 on the QSB. Below is a pic of my installation. Any advice would be appreciated!
[image]

Chris32895, 2022-12-08:
OK so I got it fixed. Looks like the second solder joint from the bottom left was not done properly. This one was ground and after identifying it with a multimeter and redoing it my xstation seems to be working. I used the attached schematic to help me out and I didn't realize that I could use pretty much any ground connection to check for continuity so I skipped that connection in my tests initially. Solder joints aren't great as I don't have a decent soldering setup but like a $15 home depot soldering gun with no heat regulator. I should have gotten something better before this mod and to anyone looking at this I highly recommend a decent rig!
[image]

ramapcsx2, 2022-12-08:
Very nice! :)

For soldering, I think the most budget friendly option is a small, adjustable soldering "station".
These start at low price points, but you can immediately tell that they're miles better than the typical "welding iron" hot sticks ;p

closed 2022-12-09 by ramapcsx2