x-station/xstation-issues #263: Juegos PAL no cargan

Potolococo, 2022-12-20:
Hola estimados,
Tengo una consola americana SCPH-1001 con XStation, pero cada vez que inicio un juego PAL, la pantalla queda en negro, a veces se escucha la intro, pero luego no se oye nada, siempre sin imagen.
Con los juegos NTSC y NTSC-J no tengo problemas, corren perfecto.
Si alguien sabe alguna solución lo agradecería mucho.

ramapcsx2, 2022-12-20:
Oye, esto será difícil de explicar a través de los traductores, pero tu consola podría necesitar otra modificación, que no corrige el reloj de la GPU en los modos PAL.

Potolococo, 2022-12-20:
> Oye, esto será diferente de explicar a los traductores, pero tu consola podría necesitar otra modificación, que no corrige el derecho de la GPU en los modos PAL.

Oh I see, I thought installing the Xstation didn't require a region unlock chip.  Before installing the xstation I was using this same console unlocked and the PAL games started without problems. 
The person who installed the xstation for me removed the unlock chip.  I don't know if it will be that.

ramapcsx2, 2022-12-20:
It does not require any more chips :)
 
It is weird that the same setup played PAL games before and had a picture, and now, only by using an xStation, it doesn't anymore.
There is no technical explanation for that, so maybe something else changed? Are the TV or upscaler still the same?

Potolococo, 2022-12-20:
> It does not require any more chips :)
> 
> It is weird that the same setup played PAL games before and had a picture, and now, only by using an xStation, it doesn't anymore. There is no technical explanation for that, so maybe something else changed? Are the TV or upscaler still the same?

I have the exact same setup and the same cables, it just changed that I now have the xstation.
I use it with a multi-region plasma monitor + an upscaler (GBS Control)
https://encrypted-tbn1.gstatic.com/shopping?q=tbn:ANd9GcR9voZGkPWuTQTdcChbhOIN73Wm28UYxr7DhjZFvnZrkBhz3KbJ0yOuQgQyUZ3HEx2myv293TGjZQw_NFd3uen-5-1mHzJtqMY7zgN2PiUsqieN2PiUs&

It's strange, before they ran all the games, NTSC, NTSC-J, PAL without problems.

ramapcsx2, 2022-12-20:
It's very strange. You can think of the xStation as a CD-ROM drive. It does not affect the PAL/NTSC video output in any way. What worked before with the real drive should work the same with xStation now.

Killerbee256, 2022-12-22:
I've tried to play a PAL game on NTSC play, it works but I get rolling distorted screen. Obviously the TV I'm using can't display a pal signal.

Potolococo, 2022-12-23:
> I've tried to play a PAL game on my NTSC PlayStation, it almost works but I get a rolling distorted screen. Obviously the TV I'm using can't display a pal signal.

After changing the resistor as directed, I tested the console using an RGB cable and with the Upscalers GBS Control, now PAL games work without any problem :)
I thank you for your great help and patience.
I hope this can be of use to someone in the future. I consider the issue resolved.

closed 2022-12-23 by Potolococo

ramapcsx2, 2022-12-28:
Cool!
It's a bit complicated, since there are multiple roadblocks. For some, it's only the resistor. For others, it's the TV, etc :p