x-station/xstation-issues #264: LEDs stopped mid blink, now black screen, no LED

bubbabobba, 2022-12-21:
Hey all,

Was working on soldering and testing an xStation install.  Wasn't working, but was getting to the Sony screen, so reflow, power on, rinse repeat.  The last time, rebooted, led came on, then abruptly stopped, black screen - and now I can't get an LED on the xStation to come on at all.  Reflowed, still no LED anymore.   I noticed a loose screw rolled under the xStation while testing it and my have contacted something on the back while it was on.  Eessshhh :/ -  could it have shorted and killed something via the back of the xStation PCB while running?  Want to know if the xStation could be DOA now before I keep trouble shooting the soldering/install.

Thanks so much!

ramapcsx2, 2022-12-21:
Okay, first check the PSX fuses, particularly PS601 (iirc), which is nearby the power connector on the PU-18 mainboard.

bubbabobba, 2022-12-22:
Bingo.  He's the dead one.  Replaced him, xStation LED came on again (woohoo), and he fried again.  Wonder if I just did a shoddy job replacing it.  Nothing on the QSB could be doing it right?  Hmm.  But yeah, I took one of the "20" fuses from a PU-18 board I had laying around.  I'm on the late PU8 board and boy is that a fun fuse to work on!  Haha.

ramapcsx2, 2022-12-22:
Well, you'll first have to find and fix that short. At least it's pretty clear what the problem is.. Good luck!

bubbabobba, 2022-12-22:
Yup!  Thanks!

closed 2022-12-22 by bubbabobba