x-station/xstation-issues #265: Strange SD card behavior

Mientje2, 2022-12-25:
I have 3 identical SD cards that I use with my Xstation (Sandisk Ultra A1).
The all have the same '00xstation' folder in the root directory
Two of them perform without problems.
It's the same SD card that gives me issues. (This SD card performed in the past without problems).
When I put this SD card in, I don't get to the white Sony logo screen.  I only get a black screen that seems to be flashing.
So my first thougt was that there is something wrong with this card but when I put it in my computer and scan it, there are no problems found.  All files are where they are supposed to be and the '00xstation' folder is intact.
(everytime I put this card in my PC it is immediately recognized). (So I guess no bad contacts and the card being brand new).

The only solution I found to use this card is doing the following.
DO THE FOLLOWING PROCEDURE AT YOUR OWN RISK!!!
- I put one of the other SD cards in and let it boot till I get to the blue xstation screen.
- open the lid and remove the sd card
- put the 'problem SD card' in
- reset
This always gets me to the blue xstation screen on the misbehaving SD card.  And it works as usual.
If I reset the playstation there are no problems I always get to the blue xstation screen.

So I suspect the loader being faulty, but I copied those files from the other working cards and it doesn't work.  I copied the loader from my PC and it doesn't work.  So all three cards have the same loader and 1 gives me problems.

Somebody have any idea on what could be causing this.
I could open the case to check out the behavior of the LED's if thats necessary.

ramapcsx2, 2022-12-28:
Hey,
I think that card could just be slightly faulty, or the formatting went bad. If you haven't tried that yet, try reformatting to exFat.
If it's still the same, then it could just be a card that has some marginal flash cells in a particularly important block.

Mientje2, 2022-12-30:
Thanks for the reply.  Would a quick format do it or better full format.  Being 400 GB this would take some time.