x-station/xstation-issues #266: Xstation doesn't register Pocketstation

Enecororo, 2022-12-28:
I just received a Pocketstation in the mail from eBay, and no matter what I do the games I try it with don't recognize the peripheral.

Is support not programmed in at all?

ramapcsx2, 2022-12-29:
Hey,
the loader can access Memory Cards, and devices that report as Memory Cards. The Pocketstation should pass as a regular Memory Card.
Maybe the code is too strict though, which would be an easy fix, I think... But I don't have one ><

Enecororo, 2022-12-29:
I'm pretty sure it recognizes the Pocketstation as a memory card, but when I go to download a game onto the pocketstation, it tells me that there isn't one inserted.

I tried with Pocket Muumuu, Pocket Digimon World, and the Japanese version of Crash 3 and none of them recognized the pocketstation for downloading games onto

ramapcsx2, 2022-12-29:
Hm, so you mean the problem is within the game already running?
XStation isn't doing anything different than a normal system at that point. I was talking about the loader Memory Card browser function only.

Enecororo, 2022-12-29:
Yes. While the game is running it doesn't recognize the pocketstation.

I honestly haven't messed around a ton with the memory card browser function in the loader, but I can't really test it with pocketstation functionality when it won't even let me download a pocketstation game.

With Pocket Digimon World, I have the thing inserted into slot 1, the time on the screen is inverted showing it recognizes that it's in a memory card slot, but the game won't recognize that I have a pocketstation inserted

ramapcsx2, 2022-12-29:
It will be a problem that would happen with a real CD, yes. Might be that you need a particular setup? Maybe the right game version?

Enecororo, 2022-12-29:
I tried it with most of the games I have that accept pocketstations. I'm completely unsure of what I could do differently. I pulled it out and put it back in and it still doesn't give me the option to download

Enecororo, 2022-12-29:
Later today maybe I'll try some stuff again. fiddle with the pocketstation more, download regular saves onto it. try downloading games again.

Are you certain that it's not an issue with the ODE? I don't really see how the disc images could be the issue, I don't think it'd be an issue with the PS1 hardware either

ramapcsx2, 2022-12-29:
Yep, 100% certain. The xStation is basically just a fancy CD. There are no patches applied to games either, that could mess with things.
Ah, but that reminds me: Please do try full booting the game. The fast boot method could have an effect.

Enecororo, 2022-12-29:
I was using NTSC-J games, no version that couldn't accept the pocketstation, I'll go try a full boot right now.

The one patch I have applied to a game is to Street Fighter Alpha 3, and all it does is makes the pocketstation option visible

Enecororo, 2022-12-29:
No changes. Street Fighter Alpha 3 tells me that there's no pocketstation inserted, Crash Bandicoot 3: Buttobi Sekai Isshu has the option grayed out

ramapcsx2, 2022-12-29:
Probably an issue with the Pocketstation, I'd say.

Enecororo, 2022-12-29:
That would be a lot harder to resolve, I paid $40 for it and it appears to be working otherwise, it even had Dokodemo Issho loaded onto it when I got it.

I did load it into the Xstation memory card manager once and it looked all corrupted, but the corruptions vanished when I unplugged it and plugged it back in

ramapcsx2, 2022-12-29:
Hm, I simply don't know these devices, and what can go wrong with using them. I can't say more, other than the xStation isn't involved.
It might need a new battery? Does it have those?

Enecororo, 2022-12-29:
It supposedly had a new battery, it uses CR 2032 batteries.

and it does turn on and show the time and stuff when it's not plugged in.

Worst case scenario maybe you can order a pocketstation as soon as you can and figure things out yourself, or wait for someone who does own one

Enecororo, 2022-12-30:
Honestly I wish the people who made the Memcard Pro also threw in Pocketstation support. Rather than having to take a crapshoot on nearly 25 year old hardware

Enecororo, 2022-12-31:
Update: The Pocketstation I have functions as a standard memory card no problem, even lets you look at the save icons while it's not plugged in to the PS1.

Which makes the issues downloading games to it all the stranger

Enecororo, 2023-01-01:
Somehow it's gotten even weirder, I tried it again tonight and it just worked. As far as I know I didn't do anything differently. (Unless saving a regular game file to it first somehow fixed it)

RetroGamer90s, 2023-01-01:
Hello
I own a pocketstation and it works just fine with Final Fantasy VIII and ape escape (Japanese version)
You need to make sure you have enough space for it to download inside the pocketstataion.
So delete all the save files inside the pocketstation and try again.
It works great on my Xstation
Hope this helps

Enecororo, 2023-01-01:
My Pocketstation was totally empty at the time it just wouldn't recognize it at all.

Only thing I can suggest is to try doing regular saves first and keep trying if anyone in the future experiences the same problem I did (Have it plugged in before you turn it on maybe)

closed 2023-01-10 by Enecororo