x-station/xstation-issues #267: Detect qsb soldering issue

loreStefani, 2023-01-02:
Hello,

Soldering newbie here :)

I have been able to successfully install the xstation on a pu-18 board but maybe I got lucky because now I'm really struggling with a late pu-8 install.

I am pretty sure it is a soldering issue because every time I resolder I get a different behavior and sometimes I am able to full boot a game and play for a few minutes until the game freezes.

The behaviors I get are the following:
1) Stuck at Sony logo (white background) indefinitely 
2) Memory card/Music player screen
3) Memory card/Music player screen and input not working at all
4) Insert disc screen and then Memory card/Music player screen
5) xstation menu frozen showing the game list (hard to tell if it is the input not working)
6) Video corruption during the Sony logo screen and then black screen
7) Audio corruption during the Sony logo screen and then no audio and no boot into the xstation 
8) As mentioned, working with a few minutes of gameplay before the game freezes and after that every reboot results in the 1) case.

I tried every time:
- switching flex cable
- switching xstation board
- switching SD card
- continuity is ok

It would be really helpful to have an idea of which qsb point can cause such behaviors if badly soldered. Any help is really appreciated. 

Thank you

ramapcsx2, 2023-01-03:
Hey,
as you said yourself, the problem is going to be with multiple bad connections.
I recommend that you reflow each solder point, while slightly pressing down on the QSB.
The goal is to bring the QSB as flush to the mainboard as possible, so that each point has a better chance of good contact.

loreStefani, 2023-01-03:
Thank you for the reply. 

Can I at least exclude some points when reflowing? For instance, can I assume that the APPL signal is correctly delivered if I get to the memcard/CD player menu? 

Thank you

ramapcsx2, 2023-01-03:
If you have a picture, and are able to run the CD Player, it means that the xStation is running and providing an APLL clock signal to the CPU, yes.
Furthermore, if the CD player sees a disk that you can play (it'll be silent), then it means most of the signals are working, too.

So if these tests pass: Do you get a "Insert Playstation Format Disk" message? This means that just the "PSNee" signals are not working, for example.

loreStefani, 2023-01-03:
Thank you for the help, I'll try again asap