x-station/xstation-issues #268: Reset issues / alternate reset point for LATE PU-8

swankerme, 2023-01-12:
I have a late model PU-8 Xstation installed + Will's PS1 in-game reset board. Every time I perform a reset (either physically or with controller button combination), the console resets (runs through BIOS animation) twice before returning to menu. This happened prior to installing Will's IGR. 

The alternative test point in the troubleshooting guide displays a later revision motherboard - is there an alternate reset test point for the PU-8 motherboard?

ramapcsx2, 2023-01-13:
Wow, that's odd.
It does reset#1, runs the whole Sony white screen, the black screen, then the loader, and then it does a reset#2?

swankerme, 2023-01-13:
No, it will run the Sony white screen, the black screen, hang for a couple seconds, then run the Sony white screen again, black screen, then the loader.

ramapcsx2, 2023-01-13:
Hm, I don't have any idea how this might happen, except for the IGR mod. The IGR could still be primed to listen for a controller reset, and when the loader comes up, it enables the pad comms and the reset might trigger then.
You say it did this before, but could you just try unplugging the controller port assembly? (This is probably the quickest way to check).

swankerme, 2023-01-13:
Are you saying I can run an additional wire from QSB IO16 to that Pin 7 to act as a secondary reset point?

The IGR board is extremely simply and only runs one wire to a resistor opposite and near that pin. It could be faulty, but it is a slim possibility. The double reset is intermittent, making this harder to troubleshoot. Removing the controller board, I discovered my physical reset button does not reset the system.

ramapcsx2, 2023-01-13:
Okay, so I don't recommend you rewire anything just yet.

It'd be best if you first tried to fix the PSU reset switch. You can put a tiny bit of isopropyl alcohol on it (let it flow into the switch), then push the reset button 50 times or more.
(Of course all with the power cable removed!)

If the reset signal doesn't want to work at all anymore, then it'd be best if you look for a replacement PSU.

And if that doesn't work, then the reset signal should probably be disconnected, PSU to mainboard, so at least the broken switch can't randomly affect it.