x-station/xstation-issues #272: [Feature Suggestion] Game name grouping

b0h, 2023-02-04:
I don't know if it's the right name to describe it, but in short I'd like the possibility for the game list to have a single entry for multi game disks, with the possibility to navigate and select a specific disk. 

Something like:

Final Fantasy VII  ---- press triangle or something ---> - Final Fantasy VII - Disc 1  - Final Fantasy VII - Disc 2 ....

In that way we can simply boot in the first disc if we simply press x on "Final Fantasy VII" , OR, if we need to boot from another disk, do the same by going in the inside menu. Of course, multi disc mode should be still possible from there.

This way it would remove some clutter from the game list, making it neater.

ramapcsx2, 2023-02-04:
This is a request that ties in with generally better multi disk game handling.
The firmware should realize when a game has several disks, and automatically build a queue.

I would still want to display every disk, as I can't know what the user intends to do :p

b0h, 2023-02-05:
What about uniforming the rom names into a standard that firmware can parse? something like GAMENAME ( DISC X ) .
That way a preliminary update can deal with grouping ( nesting the discs names inside a submenu like with the alphabetical folder layout ), just for the sake of a neater menu first. 
That way every disk is still displayed, just nested as in the folder layout.

ramapcsx2, 2023-02-06:
Let's first see how the automatic disk collection detection works out, then go from there.
Once it knows what's a set, there are more options :)

b0h, 2023-02-06:
Ok, I'll wait. I was hoping for a fast update on that direction, but I'll be a good customer and wait :)

ramapcsx2, 2023-02-06:
I'm really sorry with the features situation.
I wanted to have more nice things by now, but there's always priority on the low level actual emulation.
With now 2 years of research and refinement, and the new firmware upcoming that has all the results of that, .. soon :)

b0h, 2023-07-22:
Hello ramapcsx2, any news for this topic? Is there something planned about it?
Thank you for your patience.

ramapcsx2, 2023-07-22:
Thanks for yours!
My plans at this years start was to put out a really good xStation firmware (with the released 2.0.2), then focus on mStation work.
At the mStation loader step, there would be ample opportunity to develop the game handling, and this would then backport to xStation.
So this is still the plan now. Better game handling in general will be coming eventually, and with that, features like this here (or at least similar :) ).

b0h, 2023-07-24:
Very well, as always, i look forward to it! 
i have high hopes for the mStation , you did an amazing job up to now with the XStation. 
Thank you.