x-station/xstation-issues #274: 3 consoles have problems after the first successful launch of Xstation

M4xmark, 2023-02-06:
Hi, I recently bought 3 Xstation chips, 3 brand new 32gb microSD cards and prepared 1002, 5502 and 5552 consoles. I had no problem installing both motherboard versions, I also made sure that there is continuity between the contacts (diagrams on wiki) before first startup and cleaned everything with contact cleaner. Also putted kapton tape under lifted pads to prevent short. I formatted SD cards with recommended formatter and put update and loader (v1.6.1) to 00xstation folder and added 3 games to root.

All 3 consoles booted to xstation loader on first try. Then I scanned for games, then started playing on fast boot.

**PS version 1002**
After 5 minutes of playing Crash Bash (everything ok), the game froze in cinematic. I tried to reset the console, but then bootup screen started crushing and blinking, also sound started crushing.

**PS version 5552**
After 5 minutes of playing, I restarted the console to xstation menu and tried second game and also third game. Then I wanted to upgrade firmware, so I started updating, and it froze for like 15 minutes on the screen with text it should take around 1 minute. So a checked GitHub about help and found that update is brick-proof, so I restarted console, and it didn't boot up. Sometimes, there it is stuck at boot logo, sometimes it says insert disc. So I checked GitHub again and tried to resolder pads again. Checked continuity, cleaned contacts and still the same. I have 3 modules, so I tried them all and results are different (stuck at boot - green LED blink once at startup, insert disc - green LED blink once, then before screen change like 3 times, menu boot - green LED blinks chaotic). After like 50 tries it once boot up again to version 1.6.1, but after reset there was insert disc screen again. So again, I tried resoldering, adding more solder, checking and cleaning everything, reading every GitHub issue, no help. Tried all 3 modules, all 3 microSD cards, all formatted with recommended program, and different firmware versions.

**PS version 5502**
Similar to 1002, first few minutes playing game, then freeze. After restart, screen started blinking and sound goes bad, and then there is no screen at all.

I consider myself as good tinkerer and I think my installation was not really bad. I cleaned everything, checked continuity of all pads before startup and also all consoles worked after the first boot.

There is folder with pictures of my soldering work and videos of LED flashing
[LINK](https://mega.nz/folder/tx03TLKJ#YlKN8BSVEaZ8o1r6BDiS9Q)

ramapcsx2, 2023-02-06:
Gee, well, that's certainly not common. I can say that much.

For the consoles with erratic picture and bad sound, I would say the system has major clocking issues. 
This could be APLL related, in particular if the old oscillator was still in the system, or the wire go shorted.

The console that kind of works could be a similar issue, just from how erratic you describe it.

I'll check the videos, see if I can find an issue.

ramapcsx2, 2023-02-06:
Ah yeah, videos help.
VID_20230121_094147.mp4: This one is either shorting power somewhere or has GPU clocking issues (totally away from the xStation domain.) I suspect bad PSU or semi-short somewhere.

VID_20230125_163237.mp4: Insert Disk error is usually a QSB or flat flex, ie a signal issue. One or more signals are not reaching.
(I think all other videos are of this console?)

M4xmark, 2023-02-06:
> Ah yeah, videos help. VID_20230121_094147.mp4: This one is either shorting power somewhere or has GPU clocking issues (totally away from the xStation domain.) I suspect bad PSU or semi-short [...]

VID_20230121_094147 is old 1002 model. This flickering started after 5 minutes of gameplay, Im not sure, where could be potential short. Checked everything. Maybe PSU goes bad after few minutes?...

Other videos are models 5552 and 5502 with different xstations modules. Results are different, but when I booted up consoles for the first time, it worked fine. Maybe QSBs goes bad?! :/

I can make livestream of my work, maybe you could assist or something. I have more console to work on, so I need to upgrade them, not destroy them :/

ramapcsx2, 2023-02-06:
I often hear of installs working briefly, but if the console gets warm, they drop a signal and stop working.
It could be a cold solder joint. Good enough for passing a multimeter test, but not solid.
All the erratic xStation behaviour with the random green LED can be explained by it getting bad commands.

If you're doing many consoles, I would recommend you invest in a scope (or do you already have one?).
Scoping every signal should reveal any issues.
Of course this takes time, but it's overall quicker than rather blindly checking with multimeter and loupe.

The 1002 console has something major going on. I would try to swap the PSU on it first.
Also try and look for any stray little solder balls that could have flown between IC pins. This happens quite often.

ramapcsx2, 2023-02-10:
Hey ho, any updates? I'm available to help :)

M4xmark, 2023-02-13:
> Hey ho, any updates? I'm available to help :)

Hey, I will be working on the consoles this week. I have 3 other consoles, so I´ll try fresh installation and then I´ll let you know.

M4xmark, 2023-02-13:
> Hey ho, any updates? I'm available to help :)

Hey, I will be working on the consoles this week. I have other 3 consoles so I will try fresh installation and then I will let you know.

M4xmark, 2023-06-11:
So, I managed to install the new update and 2 consoles are already working. I couldn't get one console up and running, so I had to take another motherboard and I already have all 3 consoles with the Xstation chip working.

closed 2023-06-11 by M4xmark

ramapcsx2, 2023-06-11:
Sounds good, congrats :)
Feel free to close the issue if that solves everything :)