x-station/xstation-issues #275: Castlevania - Symphony of the Night Translation Incorrectly Scaled Font

GreatHierophant, 2023-02-07:
I have run the [translation](urlhttps://www.romhacking.net/translations/1427//url) for the Japanese version of Castlevania Symphony of the Night on my X-Station, but noticed an issue.  After you start a new game, defeat Dracula as Richter, the game gives you a screen with a text scroll.  The font on this text scroll looks ugly and incorrectly scaled.  Here is a capture of my modded PlayStation SCPH-5501 PU-18 showing the screen in question:

[image]

I do not know if this is a bug with the X-Station or the translation, but FWIW Mednafen does not show it.  This is what Mednafen shows:

[image]

This is the only text in the regular game screens I have seen which show this ugly font presentation.  The end credits might show it too.

DoctorDan1986, 2023-02-07:
It's not just you.  I noticed there are also weird gaps where whole words or phrases should be but aren't on the translation.

ramapcsx2, 2023-02-07:
It would be quite weird for this to be an xStation bug. Is there a fast way to get to this scene?

nicolasnoble, 2023-02-07:
.. you shouldn't use an emulator to validate how right or wrong something would display, *especially* when it comes to non-official patches.

GreatHierophant, 2023-02-07:
The scene is pretty quickly encountered.  Start a new game, view the opening cinematic, defeat Dracula and after that the affected scroll will appear.  You cannot die when fighting Dracula.

I realize that the emulator is not the best method to use, but Mednafen was not emulating PSX when this patch was released in 2009, yet it shows text as it was intended to be displayed.  I assume that the translation team had access to a mod-chipped PSX and tested it there prior to the patch being released, or someone else since then has tested the patch on original hardware of revisions similar to ours.  Maybe they tested it on a PS2?

ramapcsx2, 2023-02-07:
It's certainly curious, and I'll check it out :)

nicolasnoble, 2023-02-07:
It's also worth noting that mednafen infamously emulates the quirks of the first version of the PSX GPU, which only exists in very limited revisions of initial hardware, like the Japanese PU-7.

nicolasnoble, 2023-02-07:
Example of what I'm talking about: https://imgur.com/B1vt95E

ramapcsx2, 2023-02-07:
Okay, I can confirm the font scaling is bad on a PU-7 machine with xStation.
It occurs with fastboot and regular boot both.

It is most likely still a peculiar hardware glitch, but I've no idea what it is :p

GreatHierophant, 2023-02-07:
I don't suppose anyone could test this on an SPCH-5501 or similar machine?  Did everyone throw out their modchips when they liberated themselves from CD Backups? :)

nicolasnoble, 2023-02-08:
I mean, might as well ask Gemini about it...

DoctorDan1986, 2023-02-08:
> I don't suppose anyone could test this on an SPCH-5501 or similar machine? Did everyone else throw out their modchips when they liberated themselves from CD Backups like I did? :)

It definitely appears on my 5501.
[image]

GreatHierophant, 2023-02-08:
> > I don't suppose anyone could test this on an SPCH-5501 or similar machine? Did everyone else throw out their modchips when they liberated themselves from CD Backups like I did? :)
> 
> It definitely appears on my 5501.
> 

Was that by using an original laser assembly?

Gemini-Loboto3, 2023-02-08:
It's an actual bug with some hardware models where 4BPP sprites using odd U coordinates can cause distortion. Technically it should be an original bug, since the way I render text just replaces the Krom2Addr + upscaling calls with my own proportional font rendering on canvases, but I may have accidentally not realized the game was resorting to U=1 coordinates on some lines and I'm not 100% sure it's something my code triggers or if it was legit always there and not very noticeable with the bios font.

ramapcsx2, 2023-02-08:
Thanks for chiming in @Gemini-Loboto3 , and thanks for all your work, too :)
I had a this super early PU-7 model hooked up, so I tested that.
@DoctorDan1986 uses the very common PU-18 mid-generation console.
So I think this bug will be present on most consoles, if not all of them.
I wonder whether it's worth the effort to try and patch ;p

Gemini-Loboto3, 2023-02-09:
The bug does not appear at all on the console I used for testing back then, an SCPH-5502, but then again it's only in 2 hi-res screens so who cares really.

DoctorDan1986, 2023-02-09:
I think maybe the fact that there are a couple of people posting in an issues that indicates that they care.

nicolasnoble, 2023-02-09:
> I think maybe the fact that there are a couple of people posting in an issues that indicates that they care.

It's not an x-station bug, it's a bug with the cost-free translation patch you've downloaded and applied made by fans. You're not going to get this fixed that easily.

DoctorDan1986, 2023-02-09:
> 

...preeeeeeety sure that Gemini is the fellow who made the patch in the first place.  I could be wrong though.

nicolasnoble, 2023-02-09:
I never said otherwise...?