x-station/xstation-issues #276: Xstation Locked in 'Sony Logo' Screen

gspb2, 2023-02-12:
Hi! I have a Xstation installed in a SCPH-5501 (PU-18) wich I was playing normally with no problems. Suddenly while I was playing Destruction Derby Raw, the game froze. Since, when I turn on the console, the Xstation does not pass the Sony boot screen, with the lid open or closed. It doesn't even enter the "Memory card / CD Player" screen. I get a steady blue light and a blinking green light on the XStation for about 30 seconds.

Sometimes, when the console has been unused for a few hours, it enters into the "Memory card / CD Player" screen. But after some  minutes, starting again it locks on the 'Sony' screen.

I opened the console, check the soldering points myself twice, everything seems fine. Tried different SD Cards and even another PSU... Nothing worked.

I even sent the console back to the modder who done the instalation for me (I have some soldering skills, but in this case I decided to send it to a professional). He doubled checked everything and found nothing wrong.

I do not know what else to do. I can only conclude that either I have a faulty console or a faulty Xstation. Could you give me any clues?

Pictures below and videos in Google Drive:

https://drive.google.com/file/d/12ZFH57a6pnygs4Xjnr2Lpi1VnlVQwrKD/view?usp=sharing

https://drive.google.com/file/d/12Yql8q2oh428Kn2blcr5rmzV_HYHa39n/view?usp=sharing

Thanks in advance

[image]
[image]
[image]
[image]

ramapcsx2, 2023-02-12:
Hey,
from the videos and description, it seems that one of the PSX chips has died.
This condition prevents the BIOS from finishing the initialization sequence.
Meanwhile, the xStation might get bad seek commands. It is reading wildly from the SD card.

Another cause could be random small solder balls. They sometimes land between IC legs and cause a short there.
I've highlighted one of them here:
[image]

The xStation and QSB install are probably fine, judging by the SD access light and that the BIOS partially runs.

gspb2, 2023-02-12:
Thanks for the response. I recheck the board and there is no solder balls. Maybe is some dust or fiber in the picture. Just to be sure, I cleaned all the contacts and the board with IPA and no luck. I guess i have a faulty chip then... I have a replacement board, not in great condition, but it works. I will find someone to check and transfer the PSX chips. Hope it works : ).

ramapcsx2, 2023-02-13:
Yeah.. that is just really difficult to diagnose, too.
With the xStation, I think there came an increase in interest in that hardware, and now more faults begin to surface.
I've always known (well, for a couple years now) that very early SPU chips die left and right (those on PU-7), and that the CD controller on all pre combined chipset era (everything PU-18 and below) is also dodgy.
Along with that are several more random RAM failures: SPU RAM, CD decoder RAM. CPU and GPU memory seem to hold up.

That isn't to say that yours must be a chip damage problem, but it's the most likely issue, from what I can see.
Easiest way to fix it for you would be to swap the entire PSX mainboard, ie move the xStation QSB to a new board.

gspb2, 2023-02-13:
Ok, i will try that, The spare board i have is working fine, but aesthetically in not great condition overall and still have some old modchips installed. I'm afraid that extracting the QSB maybe cause some damage (if not done carefully, of course). I will send to a professional to take a look, i guess that is better. Anyway, thanks again!

gspb2, 2023-02-13:
I'm already relieved that aparently it's not the Xstation board itself and need to pay another 100 dollars, lol.

ramapcsx2, 2023-02-13:
Oh, if the xStation was faulty, we'd get it replaced.
With this particular hardware (which isn't too complex to know about most possible faults), I can often tell from afar whether it's actually broken or not. In all cases I know of so far, the seller replaced it, no questions asked.
(I can only affect their decisions on that, they are independent partners, but so far everyone agreed that it's always best to just replace :))

So ehr yeah, don't worry about that, but I agree it's a good choice to let a modder with experience swap the QSB over. That is tricky to remove.

Good luck!