x-station/xstation-issues #277: Inconsistent behaviour in NON folder-mode view

jandal55, 2023-02-16:
I am noticing that only 48 of ~250 bin/cue format games are populating the games list
I am using a number of multiple bin/cue folders - mostly from the redump romset.

Initially more of these were fine but when I cleaned up file names (removed region, languages etc), most became broken.
The method i used to rename everything was something like this:

for folder in /Volumes/X-Station/
do
	echo "sed -i s/ (USA)//g $folder/*.cue"
done

and doing multiple rename for files/folders from the MacOS GUI.

I know what youre thinking - but I have confirmed CUE files point at the correct filename, and all folder names match both BIN and CUE. Case matches too.

I initially suspected it was some kind of encoding problem or a difference in the way MacOS does carriage return or something, but have ruled this out by editing a fresh CUE file from the Redump CUE set.  I also tried clearing out both TXT files and having them re-populate from scratch to confirm no writing issues to SD. 

My Firmware version is 1.6.1 and im using a PU-18 NTSC mobo.  
SD is 128GB and in ExFat format.
Everything works correctly in the folder view.

Interestingly, If i manually add a game (eg Parodius) to both the games.txt and gamescache.txt, it works just fine, but then if i refresh my game list it will be cleared out again.

Ive included my .txt files, as well as example CUEs (renamed to TXT - Alundra working, Parodius not working) and md5 hashes for BIN files, and directory listings. Hopefully this can rule out some kind of metadata issue.

[Parodius.TXT](https://github.com/x-station/xstation-issues/files/10759853/Parodius.TXT)
[Alundra.TXT](https://github.com/x-station/xstation-issues/files/10759854/Alundra.TXT)
[games.txt](https://github.com/x-station/xstation-issues/files/10759952/games.txt)
[gamescache.txt](https://github.com/x-station/xstation-issues/files/10759953/gamescache.txt)
[history.txt](https://github.com/x-station/xstation-issues/files/10759954/history.txt)
[config.txt](https://github.com/x-station/xstation-issues/files/10759973/config.txt)

> % ls -la /Volumes/XStation/Alundra
total 1173504
drwxrwxrwx  1 user  group     131072  9 May  2020 .
drwxrwxrwx  1 user  group     131072  1 Jan  1980 ..
-rwxrwxrwx  1 user  group  600359760  9 Oct  2019 Alundra.bin
-rwxrwxrwx  1 user  group         70  9 Nov 11:27 Alundra.cue`

> % ls -la /Volumes/XStation/Parodius
total 1021952
drwxrwxrwx  1 user  group     131072 10 Feb 20:13 .
drwxrwxrwx  1 user  group     131072  1 Jan  1980 ..
-rwxrwxrwx  1 user  group  522741408 24 Dec  1996 Parodius.bin
-rwxrwxrwx  1 user  group         74 16 Feb 12:04 Parodius.cue

> % md5 /Volumes/XStation/Parodius/*
MD5 (/Volumes/XStation/Parodius/Parodius.bin) = 6876275f34e437a99d1bc07d176dcd38
MD5 (/Volumes/XStation/Parodius/Parodius.cue) = b535f3c15fca83335c74334fa64295fe

> % md5 /Volumes/XStation/Alundra/*
MD5 (/Volumes/XStation/Alundra/Alundra.bin) = ff911668c363132913f5d06f31780340
MD5 (/Volumes/XStation/Alundra/Alundra.cue) = 45292873c556298fecab5e0b73365c6a

ramapcsx2, 2023-02-17:
Hey,
I'm sorry there's issues. Let's try to find out what goes wrong :)

So the file scanner totally ignores .cue files. Anything about them doesn't matter.
The scanner iterates over all directories and adds all .bin files by extension. That's all it does (hence the scan speed).

What can go wrong though is those awful MacOS metadata folders and files.
I try to skip those that I know of, but every now and then, there seems to be a new edge case.

I think you'll find that if you were to clean the SD card and copy your games using Windows, it would work.
As an alternative, maybe there is something you can change on your Mac that fixes the metadata creation.
Simply not hiding metadata files, so you can manually delete them, should also work.

ramapcsx2, 2023-02-17:
It would also be best if you could undo all the file renaming, as there's some risk that it's not to expected standards.

jandal55, 2023-02-17:
Hi Rama - Thanks for the quick reply.
This context on *how* the list populates is useful - i saw another closed ticket that said it does some filtering based on parenthesis and was worried.

Your suggestion helped me to solve the issue though:

1) I disabled indexing on mac but that didnt really help me.
2) Inserted the SD into a PC
3) Noted that MacOS creates folders which CANT be delete in MacOS -.fseventsd, .Spotlight-V100 and .Trashes - and deleted these in windows (this likely clears up a minor issue in folder view where a "?" empty folder is displayed)
4) Deleted any .DS_Store files - there were about 3 of these
5) Searched for all ._ named files and deleted them - there were about 6 of these - 2 in game folders (including parodius) but i think more pertinently, there were some created for 
._games.txt
._gamescache.txt
._history.txt
._config.txt

After the above steps, all games are showing up correctly, and i get the same game count in folder and non-folder view. Thankfully i didnt need to re-format or re-transfer games.

Hoping the above steps can help you with any filtering x-station might do - if you want me to go back and break it to determine which exact step fixed the issue, just let me know - my suspicion is step 3 or the 4 files in step 5.

ramapcsx2, 2023-02-17:
Cheers :)
The file walker probably cancels out in one of these metadata folders, for one reason or another.
When it didn't find any more files or folders, that top level folder "is done".
It's just one big headache with Apples metadata files on external storage media..