x-station/xstation-issues #278: PU-18 No video/Power LED flickers and then stops completely

jonnyfire, 2023-02-21:
Wondering if this x-station is just fried. after installing it was having issues reading the sd card so I took it apart and noticed I missed a pad however after putting it back together it stopped working entirely. It outputs no video and the power LED will either flash for a second or if I hold the lid switch will start to flicker for 5 seconds afterwards it wont light up no matter what. I've gone over and tested all the joints and fuses for continuity and voltage and everything is in order. I really don't understand what's up

[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2023-02-21:
The power LED flickering like that is a very bad sign of (most likely) a short / almost short on the 8V rail of the system.
Please check the fuse PS601 on the PSX first for continuity, then for resistance. It should read almost 0 ohms.
Next, something is causing that short. You need to find where it might happen and fix it.

ramapcsx2, 2023-02-21:
The QSB is missing a section? That section used to have the entire grounding, so that would be important :p

ramapcsx2, 2023-02-22:
So as a folllow up, without ground in the xStation system, power will find a return path elsewhere. It'll flow into chips, doing damage all around. This is the problem, and I hope the rest still works after fixing that :p

jonnyfire, 2023-02-23:
There is still a ground pad on the board. Its the bottom of the three under IO4. I don't know how important it is too have 2 extra connections but all the continuity is fine.

ramapcsx2, 2023-02-23:
Not enough, that is a connection that drags down a signal to the decoder. It's not meant to be the system ground.
You really need to restore one good ground, for example using a short, not too thin wire, soldered to the flat flex connector and the nearest good ground.
Also, there is one more connection at the end of the broken off "arm", that needs to be grounded.

Once you have all that, if the blue LED still flickers, immediately turn the system off and look for more problems.
There could be shorts, etc, something strong enough to drop an 8V line so low that the 3.3V regulator doesn't work anymore.

jonnyfire, 2023-02-23:
Oh it really was that simple. Would have thought to do that if there was some broken pcb in the bag but that is just how I received it. thank you very much for you time!

closed 2023-02-23 by jonnyfire

ramapcsx2, 2023-02-23:
Good thing the PSX can be sturdy when it wants to :)