x-station/xstation-issues #279: Black Screen / Power Led ON / No activity Led

vdorza, 2023-02-23:
Hello all. As the title states, after i installed my xstation on a pu18 mb i got a black screen. I checked all the solder points, and tested continuity.  All checked up. Still black screen. I blamed myself, maybe i did something wrong. I uninstalled the qsb and tried another pu18 motherboard. Again, black screen, no activity led.
The only thing that came to my mind was, buy another xstation, which i did. Once i received it i installed again on the same motherboard that had the qsb installed, and voila. It worked first time. This leds me to believe that the first unit i got is somewhat defective. Thing is i want to try to fix it. I have my ps already moded, but i do not want the first unit to go to waste if maybe i can do something. Maybe i can replace the esp module? If so, can someone advise? 
Thnks

ramapcsx2, 2023-02-23:
Hey,
I need to know what the bad one does. Do any of the LEDs light up? Is there any visible damage?
The ESP modules can't be replaced in the field. You'd send a defective unit back to the seller for a replacement.

vdorza, 2023-02-23:
Thnks for you quick reply. The power light (blue) turns on and keeps turned as long as the console is on. The green activity does not turn on nor blinks.
[image]
[image]

ramapcsx2, 2023-02-23:
Okay, it could have simply died. Maybe it's best to have this one swapped..