x-station/xstation-issues #281: WipEout 3 Special Edition(PAL)/Freeze on first loading screen on PAL PU-18 Console

Verdantree, 2023-03-04:
Attempted to play WipEout 3 (NTSC) on my PAL PS1, but it froze at the start line because of the region protection issue. 
Loaded up the PAL version of WipEout 3 Special Edition instead thinking it'd work without issue. PAL console, PAL Mechacon, no problem right?

However with this version of the game (using the Repack) no matter what I seem to do it freezes at the first 'LOADING_' screen right after the splash screens, the '_' will flash twice, stutter, flash a third time and then freeze. In addition, if I have a Duashock plugged in it starts vibrating indefinitely until I power off/reset the console and occasionally the top of the screen will 'error', and a black square with red and green pixels will crawl across the screen, though I can't seem to recreate that issue reliably.

I can't seem to find anything online suggesting this is an existing issue and an emulator runs the repack without issue.

Any thoughts?

List of solutions I've tried:
Different Controller (OEM Dualshock, OEM Dualshock 2, Regular PS1 OEM Controller)
Repack hosted on a different site (I assume they're the same rip)
Merging the multitrack .bin into a single .bin
patching the single .bin file with the NTSC/PAL Selector Patch booting in PAL mode and NTSC
Fast Load and Full Load
Skip All Splash Screens
Let all Splash Screens + Intro Movie play out without skipping any.

ramapcsx2, 2023-03-04:
Hey,
have you tried the regular release?

Verdantree, 2023-03-04:
Just tried it out, Unpatched the regular PAL version freezes at the start line like the NTSC version. Patched it works as it should. Would be nice to get the other version working though as it has more content.

ramapcsx2, 2023-03-06:
Okay, so this could be a broken patched image that needs some kind of fixing.
There is repair code in xStation for some common fan patch style errors, but it's not full coverage.
Please try something like ECC Regen on the image, see whether it finds / corrects any errors: https://www.romhacking.net/utilities/1550/
In addition to that, the copy protection is very much customized in this title, and I'm not sure what a fan modification might do to it.
Can you check whether the image is supposed to run fine on original hardware?

Verdantree, 2023-03-07:
Ok you've lost me a bit here. I tried putting an unpatched copy of the repack of Wipeout 3 Special Edition into ECC Regen and it hasn't kicked up any errors. 

I'm not entirely sure how to find out if it's meant to run on original hardware.

ramapcsx2, 2023-03-07:
:)
Sorry about that :p
There is often a readme.txt along with a community fan patch, or maybe a website. It'll be good to know what they write about running on hardware.

ramapcsx2, 2023-03-07:
Well, okay, the problem is going to be that the patch broke some code in the game. I would have to speculate, but maybe it works on some hardware (that they tested), and but not others..
It's possible then that some old forum posts about this problem exist (google game + special edition + loading).