x-station/xstation-issues #282: Compatibility issues with dance dance revolution

Kingbrian16, 2023-03-08:
Hello, 
There seems to be a compatibility issue with dance dance revolution series 1-5. It will start up and work  until you select the the main game,  it will then display now loading and remains there without any progression. This happens on Fast or Full boot on current firmware.

ramapcsx2, 2023-03-08:
Yep, other users found it a few hours earlier. It's a regression in 2.0.2.
I found the cause of it and there is a fix, but please give me a while to make it a worthwhile update.
In the meantime, if you need these games to run, please revert to 1.6.1.

The problem is detailed over on Discord. Basically, it is a very peculiar form of tamper detection.

Kingbrian16, 2023-03-09:
Thank you so much, I’ll wait for the update!

closed 2023-03-09 by Kingbrian16