x-station/xstation-issues #283: Updated to 2.02 and all I see is the Playstation Logo

terenctbrobots, 2023-03-10:
So I updated to 2.02 today and everything seem to go ok. After the update, the console didn't reboot automatically(I left it for 30 mins). I did a reset..At first boot the system when straight to the memory manager and then booted into xstation. After a few resets, it now just boots to the Playstation logo and freezes there...

I ran a jumper the pin that is supposed to do the reset to no avail. 

Any idea what to do next beside go through each pin on the QSB and reflow it?

ramapcsx2, 2023-03-10:
Hm, can you say whether you had been using the console fine before, and it immediately changed behaviour after the update, or was it unused for a while, and the problems just started alongside the update?

For debugging, it sounds a bit like the PSNee signal isn't coming through.
That would be the left side of the QSB.

terenctbrobots, 2023-03-11:
It's a PU-18 board if that's any help. I just went about reflowing the qsb solder points and now it's booting 1 out 8 times or so.

Which pin on the QSB flex is the PSNee signals? maybe I can pay better attention to those. 

It was booting fine before the update and playing games. Althrough one thing I noticed is that audio seems to not work now. 

Do you think it's bad caps? I recaped about 1/3 of the board already. I may just keep going 2 at a time. Sigh..worse comes to worse I have to pick up new playstation when I go to Japan.

ramapcsx2, 2023-03-11:
No, it's not going to be caps, but very likely a partial contact on the QSB solder points or the flat flex (both ends).
The update can have such an effect, though unlikely, whenever I change drive strengths of signals ("push through weak connections"), but this update didn't modify them, iirc.
You can of course try reverting to 1.6.1, and if it fixes it, that will be interesting to know.

The fact that it sometimes works means that the console and xStation are most likely fine, but as mentioned, a signal isn't fully connecting.
Look at all the left side of the QSB signals for PsNee, though missing music indicates a different problem. It's a bit hard to tell from afar.

terenctbrobots, 2023-03-12:
Sooo...after a lot of patience I dowgraded to 1.6.1 and now it boots all the time. 

I did do a day reflowing but that didn't really change anything till I downgraded. 

For now, I am going to leave things as it is.

Let me know if you need any more info though

ramapcsx2, 2023-03-12:
It's a curious case for sure.
The note about missing music is a hint. Does that still occur? (And if so, are you game rips okay?)

terenctbrobots, 2023-03-13:
No,the music and sound are just the loose audio leads.  The only things not working right now are:
- Power on issues when the case is fully assembled. Need to check the power board and console/power cable
- Full boot doesn't work. The reset line has 0.04 ohm reading on it. Hopefully running a bodge will fix it

Could it be any timing issues? Anything to do with the PS1 being a Japanese NTSC model? I know that Japanese PS1's have region locks.

ramapcsx2, 2023-03-13:
Heh, somehow the notion spread that NTSC-J consoles cause more issues / are less compatible, but they're just a bit different.
Outside of 2-3 select titles, they behave the same as the other regions with xStation :)

Both listed issues are likely causes for the 2.0.2 troubles. If the reset signal state isn't quite clear, the effects will be quite random.
One maintenance procedure for the PSU reset switch is dropping a single drop of deoxit/contact cleaner into the switch, then pressing it many times to clear all debris (100 x or so).
Another check is for the power button. If the console is off, and just slightly touching the power button sometimes turns the LED on (console on), then the shell sits too close to the power switch.

terenctbrobots, 2023-03-14:
Just following up for your reference:
- I fixed the wierd issue with the console not powering up properly. The powersupply I use is one of those new ones you can get from AliExpress with a 12V DC input since my voltage is 220v here. The problem was that the reset button was riding the power supply switch causing reset. I hade to file the reset switch down a little to make sure it worked.
- I am still having issues with a full-boot(and most likely boot on update). The full boot resets the console, but it goes into an indeterminate state. If you hit the reset quick, it will boot the game. Haven't done any microcoding since University, but maybe you could increase the length of the reset pulse? I will try that bodge wire then I have time, but I suspect it won't do any good. Maybe the pull up resistor is borked...

ramapcsx2, 2023-03-14:
Great, it was the reset button thing :p
Knowing that you use these PSUs also helps. The clones leave out the reset circuit that is normally required.
Without it, I'm not sure how the system behaves.
If you still have it, can you check with the original PSU?

terenctbrobots, 2023-03-16:
So. I finally got round to swapping the power supply with a known good one. Now the full boot works. I'm not brave enough yet to do a firmware upgrade though. So for all those who bought those Ali Express power boards...;P

terenctbrobots, 2023-08-09:
So I decided to 'upgrade' to 2.0.2 yesterday and my unit which was running fine on 1.6.1 had trouble getting to the PS1 logo screen and continuing to the xstation menu. It would load intermittently but to get it to get into xstation proper, I had to create an empty card which I loaded with the old firmware. I then downgraded it and it works again. So I definately think something in 2.0.2 is causing my boot issues.

ramapcsx2, 2023-08-09:
Hey, while sounds plausible, there is still a factor with your hardware at play. 
I don't know whether it still has to do with the power supply, maybe the reset line, or something else...
The fact it works on the older firmware could have to do with it having slightly different timing, where it might power through the hardware problem..