x-station/xstation-issues #284: Problem updating firmware.

DiegoGonzalezSalgado, 2023-03-18:
I was with my recently installed Xstation, which seemed to have problems reading the sd or with the white cable, when I was finally able to access xstation, I decided to install the new firmware, it got stuck in loading and did not progress, now it does not access at all way to Xstation, any ideas? thanks for your time

ramapcsx2, 2023-03-18:
Hey,
both issues are probable the same thing.
If you mean an SD card extender, then please try to run it without it. If that works then, you can update the xStation, and then it should work with extender installed.

DiegoGonzalezSalgado, 2023-03-18:
I mean that in the middle of an update the process stopped, I use a normal SD, without an extender or anything, the truth is that I have tried everything already, I was very excited because it worked before updating.

ramapcsx2, 2023-03-18:
An update is hard to fail, but if it does, it will automatically unroll (rather, it doesn't get applied until it has been verified).
So a failed update does nothing, in effect.
I think you've lost some signal to the xStation, but it's hard to tell with the information I have.

DiegoGonzalezSalgado, 2023-03-18:
It's a great ray of hope, since it seemed to me that I tried to boot a little while ago, maybe it's the white strip "AWM 20634 80C 60v vw-1" I don't know how to refer to it, but I got to the sony logo, ("Sony Computer Entertainment Inc .) Thank you very much ramappcsx2 for the information! I'll keep trying.

DiegoGonzalezSalgado, 2023-03-20:
Well the problem was the SCPH-1002 Motherboard i have, something was wrong, i get a SCPH-5xxx and is working properly at 1st attempt, thank you for your time!!