x-station/xstation-issues #285: Can't boot into xStation, but looks like SD card is being read? (no green blink + debug vids)

icd2k3, 2023-03-19:
Hey all! I would be eternally grateful for any help or guidance - I been hankering to play some MGS...

I'm really at a loss here! I've seen other issues with the blinking green light to indicate an SD card that can't be mounted. Mine is not blinking, but alas, I am still booted to bios.

Hoping someone might be able to help steer me in the right direction for trouble shooting this issue.

**Pre-Mod**
This SCPH-5501 worked fine pre-mod. Was able to play GT2 and left it running for a couple hours in the optical drive without issue.

**Post-Mod**
Console turns on with video and sound, the blue and green lights are active and the green light is NOT blinking like I've seen in the SD card mount issues (see video!). A config file is created on both cards, and the CD player can "play".

**Things tried**
Tried 2 SD cards (a Sandisk Ultra 512gb and a TeamGroup Go 256gb both formatted as per the user guide. With 00xstation folder and latest firmware and launcher 2.02. I also did a full disk clean of both and a full format). Tried both cards directly slotted into the xStation, and both cards in a Lazer Bear mount.

Is there any specific places I should look that might be causing this particular issue? I'm wondering if there's certain joints or something I should come back in and re-touch. 
 
Videos:
- [Boot test](https://www.dropbox.com/s/d2ta93o6mgszzao/BOOT_TEST.mov?dl=0) trying to boot into xstation. Shows boot to bios and not-blinking, but not-quite-steady green light
- [Boot test with optical drive](https://www.dropbox.com/s/0vxsakpc3mhvjsg/BOOT_TEST_WITH_OPTICAL_DRIVE.mov?dl=0) Same as above, but with the optical drive plugged in to show it's spin up/down behavior
- [CD player test](https://www.dropbox.com/s/vu0f3s4fcmc83zy/CD_PLAYER_TEST.mov?dl=0)

Images (high res)
- [APLL](https://www.dropbox.com/s/hqm85yjgvqn1mo7/Photo%20Mar%2018%202023%2C%204%2033%2016%20PM.jpg?raw=1)
- [QSB 1](https://www.dropbox.com/s/14c6o6rxfwejyqv/Photo%20Mar%2018%202023%2C%204%2033%2054%20PM.jpg?raw=1)
- [QSB 2](https://www.dropbox.com/s/07yl4y28wgdoolj/Photo%20Mar%2018%202023%2C%204%2033%2049%20PM.jpg?raw=1)
- [Pins 1](https://www.dropbox.com/s/vqvefdsac7ohjlg/Photo%20Mar%2018%202023%2C%204%2034%2053%20PM.jpg?raw=1)
- [Pins 2](https://www.dropbox.com/s/8ja7tuxjf3d5lcu/Photo%20Mar%2018%202023%2C%204%2035%2009%20PM.jpg?raw=1)

ramapcsx2, 2023-03-19:
Hey,
thanks for the videos. They explain the problem very well to me :)

At this point, the SD card enters linear reading. I know that this is wrong right away, and I can tell that there is an installation error:
[image]

The solder job looks okay from afar. Please measure the region around IO4 though.
You want to see mega ohms between each of the 3 points. One point should be grounded.

ramapcsx2, 2023-03-19:
In the IO4 region, particularly these 2 look shorted:
[image]

icd2k3, 2023-03-19:
@ramapcsx2 thank you so much for the speedy reply! 

> Please measure the region around IO4 though. You want to see mega ohms between each of the 3 points. One point should be grounded.

I will order a multimeter tomorrow (can you tell I'm a bit new at this?).... I will also be googling how to use a multimeter 😅 ... assume I will be mostly referring to [this image](https://user-images.githubusercontent.com/6278722/112033652-f08d7380-8b3d-11eb-9124-131df4cb646f.png) in the readme? Any beginner multimeter tips appreciated!

While I'm waiting for it to arrive, I'll attempt to visually clean up those 2 points you highlighted and see if it helps.

Again, many thanks for the speedy reply @ramapcsx2! Thought I might be shooting in the dark here and truly appreciate you having a looksee.

icd2k3, 2023-03-19:
@ramapcsx2 nailed it!

Didn't even have to wait for my multimeter to be delivered, I just cleaned up that joint and it boots into xStation no problem now - thanks!!!

ramapcsx2, 2023-03-19:
Awesome, glad I could help :)

icd2k3, 2023-03-19:
@ramapcsx2 well onto the next thing - I can get into xStation now, but it struggles to launch games and sometimes freezes on startup ... looks like I might need the multimeter after all ... at least I'm making progress

icd2k3, 2023-03-19:
Re-opening (sorry) was able to get into xStation once, but now it keeps opening bios instead :( looks like I still have a bad connection or something somewhere - I have a multimeter coming later today, but not 100% sure what #s I should be looking for.

icd2k3, 2023-03-19:
Looks to be some kind of intermittent issue with lid detection... is there any way to fix it so the lid always registers as closed? or is that a bad idea for other reasons?

right now I'm wedging a folded up little piece of paper in the back so it registers the lid is closed, not the most elegant solution lol

ramapcsx2, 2023-03-19:
Hm, if you take it apart and inspect carefully, maybe you can see where the mechanism looses a couple millimetres.
The lid switch is a 4 point solder job. If you can heat up all 4 without destroying the switch, you can push it outward a little.

closed 2023-03-19 by icd2k3

icd2k3, 2023-03-19:
Ended up just putting a couple layers of thick mounting tape on the top of the switch and on the latch... I have a replacement shell coming soon from Muramasa so I'm hoping my current shell is just a bit warped or the pin is worn out or something

anyways, thanks for your help! xStation is awesome (been playing Ape Escape with my 5yo all afternoon)