x-station/xstation-issues #286: Works intermittently, but trouble booting into xstation (continuity tested - maybe bad lid detector?)

icd2k3, 2023-03-21:
Sorry to bother again, but I'm still having some issues I'm hoping for some insight on!

previous issue: https://github.com/x-station/xstation-issues/issues/285 (this thread has hi-res pics of my solder job).

So based on the suggestions in this initial thread, I got it up and running. It was working fine for a day, but now when I power on the console it consistently boots me into the bios MC manager / CD player.

**General behavior every power-on now is:**
- It will boot into xStation and I can play games (🥳 )
- It will turn on and freeze on the "PlayStation" black logo screen *
- It will turn on and boot to bios (this seems to be the most frequent now) *

> * For these issues, the power light on the xstation will be on, but the access light will only quickly blink one time, then off.

**Stuff done:**
- Invested in a multimeter and confirmed continuity to all the points in the readme diagram
- Used multimeter to inspect for shorts in the QSB and in the lifted pins/pads
- 2 different micro sd cards and completely diskpart cleaned and overwrite formatted one of them

Could this maybe be a dead or faulty lid detection switch or something else? I ask because while debugging, I noticed I needed to hold down the lid detection switch manually for it to boot into xstation. I installed a spring to give the lid a little extra leverage, and that seemed to do the trick... until today.

Now, even when I manually hold down the detection switch I still get consistently booted into the MC manager / CD player which is worrying...

I ordered a replacement lid detection switch from console5 already, but is there anything else I should look into to debug this kind of issue?

Thanks!

icd2k3, 2023-03-21:
thank you @ramapcsx2 

I get about 3.5MO resistance to ground on `14`... but on `4` there is 0 resistance to ground - so it seems like the problem is there maybe?

Neither 14 or 4 beep during a continuity test to ground

ramapcsx2, 2023-03-21:
Hm, it could just mean "open line", which is infinity ohms. This is a reading where the multimeter just can't give a valid value anymore, and it means it's okay.
Resistance between 4 and 14 is also good, I assume?
If so please test point 20 against ground next (should be high reading (or open line)).
Next, please make sure that point 19 is actually reaching where it should (common error spot)

icd2k3, 2023-03-21:
> Resistance between 4 and 14 is also good, I assume?

yea ~10MO

> If so please test point 20 against ground next (should be high reading (or open line)).

Getting 06.52MO reading

> Yep, continuity test beeps

I decided just to re-flow point #4 in case and now I'm getting a resistance reading of 3.5mo resistance to ground... so maybe?

icd2k3, 2023-03-21:
So I don't want to celebrate too soon, but it appears to be working now - I'll test it a bunch tonight and see if I start getting booted back to bios or stuck PS logo, but I just tried 3 times and booted straight into xstation each time

I'll turn it off/on like 25 times tonight to make sure the behavior sticks :D but looks like _maybe_ `4` resistance to ground was the issue

icd2k3, 2023-03-21:
@ramapcsx2 got it, so soak up as much solder as I can from ground _and_ 4, then re-solder ground again (both with minimal solder)

I'll do that tonight... I guess worse case scenario I might have to take out the QSB to be able to clean under it and maybe get a replacement?

icd2k3, 2023-03-22:
Ok, I was able to clean a good bit of it out using solder wick & a sucker

[image]

then added a little bit of solder to ground and a little to 4 as recommended

[image]

then I ran all the continuity, short, and resistance checks on the 3 points.

...

As of now everything looks good! Booted straight into xStation 3 in a row, then played a few different games for 1.5hrs and tried a few more times without fail. Not only that, but everything is loading much faster and I have no more FMV stutter issues. If it starts to have these problems again at least I'll know where culprit probably is, and how to fix it (I'd remove the QSB, clean under it, and probably replace it with a new one).

Out of curiosity, what is it about `#4` that would cause these symptoms? (boot to bios, freeze on PS logo) - I'm just wondering what it connects to or what it controls.

As a beginner modder, I learned so much and actually had a ton of fun with this mod despite the problems I ran into.

@ramapcsx2 **I could not have done it without your help!** Don't know how to thank you - if there's a way, let me know!

closed 2023-03-22 by icd2k3