x-station/xstation-issues #288: No luck with xstation install.

SpyroPSX, 2023-04-24:
From Australia.
Purchased xStation from Dragonbox.
Have attempted install on an SCPH-5502 and SCPH-5500. Same issue, get the mem card, music player screen. xStation power light is on and green light flashes once on boot with SD card installed. If no SD card green light flashes constantly. Have checked my test points and confirmed continuity as well as looking for bridge's between close solder points. Any suggestions?

ramapcsx2, 2023-04-24:
Hey,
you have the default signal error issue, I'm afraid. 
It means that one or more signals are not actually connected fully, or they are bridging each other, or any other signalling issue like that.
Please measure groups of packed QSB solder points against each other and look for the Ohms reading.
If you spot a pair that read less that a few hundred kOhm (and they are not supposed to be grounded or pulled up (reset line)), then those are suspect.

SpyroPSX, 2023-04-24:
Hi Rama,
Cheers for your swift reply mate, I will check that out as soon as I get home and will get back to you. Thanks! :)

SpyroPSX, 2023-04-25:
Hi Rama,
Apologies for the delayed response, I wanted to make sure I tested everything properly before responding. Still no luck unfortunately, just getting a single green access flash on power up and a solid blue power led. The white Sony logo definitely hangs for longer with the lid closed like its trying to do something before going to mem card cd player screen.

I tested all pins resistance to GND and found pins 1, 11 & 18 GND as expected. Pin 2 measured approx 50k and pin 17 approx 8.4k.
Pin 4, 14 and 15 were approx 4M to GND.
All other pins measured open.
I found the same when I measured between solder joints on QSB, found no 2 points that actually measured a resistance reading. I dont know if that's to be expected or not?
I reflowed all the same and retested getting the same results as above. 

I am very much assuming I have made a mistake I just can't for the life of me figure out where or what it is. Any guidance is greatly appreciated.

SpyroPSX, 2023-04-25:
I also reconfirmed all my test points as per the guide and can confirm continuity between the pins behind the flex connector and the test points.

ramapcsx2, 2023-04-25:
Thanks for the good report :)
It tells me that your install seems to be fine, as only the expected signals show a low Ohms reading (reset on 17 and 8V power on 2).

Please take a good look at 19 next, which sometimes isn't fully connecting to the (square) Mechacon IC, on that pin where the "19" label is.

You may want to enter the PSX CDDA player to confirm whether any CD action works at all.
Start the console with the CD lid opened for this, enter the music player, close the lid and it should find a CD to play.
It should be possible to play the CD, with the time counter advancing normally.

SpyroPSX, 2023-04-26:
Hi Rama,
Managed to verify the connection on pin 19. No issues there. However I followed your instructions above to verify CD activity and I have absolutely no CD response at all. It doesn't even try and read a music CD. 

Just so we are on the same page I assembled the PS1 with the xStation connected as well as the CD drive and inserted a music CD after booting up with the lid open and entering the music player at which point I closed the lid. To verify what was happening I also tried with no lid and just pushing the lid button down once in music player to see if the drive did anything and no spinning or movement. Out of curiosity I tried booting without the xStation connected but this resulted in nothing but a black screen but I assume this is normal due to the modifications.

ramapcsx2, 2023-04-26:
Okay, it could be related to signal 20, which is the drive "limit switch". It tells the system that the read head is in innermost position, and if this is "stuck" there, possibly due to a bad connection, the system will never try to init the CD.
You could check the voltage level on this signal when pushing down the lid switch, at which point it should visibly toggle between high and low. This toggling has to reach the Mechacon.

L10N37, 2023-04-30:
I am based in Australia if you need a modder. VajskiDs consoles.

Last attempted install, i bought it back to life but the console had these problems

1: Took an extremely long time to boot
2: Would not go into the Xstation Loader
3: Had no audio

Obviously issues caused by the bad installation. I confirmed the installation as good on every point and had to run a bodge on 2 broken ones. 

I ended up having to replace the board, I've got a couple PAL systems left but have stacks of JP Consoles, but i pay about 100-120 for each, i got stung with way more taxes and shipping than expected when i bulk purchased a tonne of Xstation compatible systems. I install and diagnose issues on these systems often and I know for a fact that a dead PS1 board is not worth the time or effort repairing or attempting to repair unless it's just a fuse or something simple. 

I am just doing some quick adjustments to things and about to lodge a strange issue myself (it's not major, but good information to share)