x-station/xstation-issues #289: Xstation won't boot after failed update.

Ericonzeta, 2023-04-27:
I was in the process of updating my station, but I started the process with the lid open. I waited a long time, but it hung on the update screen. After a reset, it wouldn't boot into the xstation menu anymore. I have a solid blue light and the green one blinks when turned on. Oddly enough, if I unplug the PS1 for a moment and I try to turn it on, it boots into the memory card and CD menu. But every other time, it's stuck on the White screen with the sony logo. Is there anyway to reflash it?

Ericonzeta, 2023-04-27:
It does create the config file on the SD, so i think it's alive still.

ramapcsx2, 2023-04-27:
Hey, the update went through normally. The updater can't show you a finish message, because the CD was missing.
So I believe your console is now running the new code, which somehow doesn't like something on it.

First test: Do you use an SD card extender? If so, try without.

Ericonzeta, 2023-04-27:
I'm not using an extender, just the SD. I've even reformatted it and tried
other firmware versions on it. But no luck. Should I try a different SD
card? If so, what format?

Ericonzeta, 2023-04-27:
Thanks for getting back to me, do you have any other suggestions?

ramapcsx2, 2023-04-27:
You also asked on Discord, right? Let's chat there, it's easier :)

ramapcsx2, 2023-04-30:
Closing this issue, as there's contact on Discord. 
The problem so far looks like a bad console, PU-8 chip damage.

closed 2023-04-30 by ramapcsx2