x-station/xstation-issues #290: JP BIOS verse EU BIOS, one works with a certain SD card, the other doesn't

L10N37, 2023-04-30:
Hi Rama, 

I've had quite a few of these systems in lately , as well as orders of premods. All is usually fine but I've encountered a strange issue i thought was worth making public as it could cause headaches for some people.

For starters, I made sure the systems were completely even in most regards

1: Latest FW from Feb
2: Both PU18

Not everyone has multiple Xstation PCB's and systems on hand for testing, so worth sharing for sure.

So I have a brand new 512Gb Sandisk card from a customer. Using Rufus the card was formatted as FAT32 and filled up - 461Gb, alphabetical 'A to Z' folders including a '#' folder which has never caused issues.

Card didn't work, not recognised by Xstation. Just green flashing. I know exactly when the lights need to blink for a successful black screen load (as sometimes the ribbon cable will allow the CPU clock signal through, and thus the system boots, but it won't have enough contacts successfully bridging for the SD card to load).

OK

I decided to start from scratch, wipe the card, used Rufus to format again but this time as eXfat, no deal. 

OK, I have a system with a JP BIOS on hand, I put the card in that - it works perfectly. I didn't even need to update the games list (new feature?)

Loaded a game up. Zero issues.

Now the other PAL BIOS system works fine with my personal 400Gb Sandisk. 

- In summary -
We have 2 identical systems - bar one being an EU BIOS and one being a JP BIOS.
The JP system has no issues with the 512Gb Sandisk card
The EU system just blinks a green light non stop

They both however are fine with the 400Gb card.

I've had similar issue in the past where i just decided to use the card that the Xstation wouldn't read in my fenrir instead, and replaced the Xstation one with a card that read. It's only tonight I find that it was particular to the system.

ramapcsx2, 2023-05-01:
I'd be reaching here to come up with possible explanations. .. Yeah, it wouldn't be power supplies or other external factors.. I draw a blank :p

L10N37, 2023-05-01:
Very strange.
He's agreed to switch out the card for another brand of which I bought a few on sale and I'll prep that and hope for the best. Weird that one system (tried two xstation PCBs) refuses to read that particular card and the JP system has no issues. The JP system read it with a recapped 240v supply and a recapped JP supply (it's actually going to Japan today)

L10N37, 2023-05-02:
Just to make public, Samsung EVO PLUS 512GB (white) micro SD, exact same 1:1 as the Sandisk works.
I've had 100% success with every EVO plus but hit and miss with Sandisk. The last 512Gb sandisk is in my fenrir as I used to jump on which ever was on special. At the moment memory is hitting all time lows so i grabbed 4x of these cards for what 2 would cost 6 to 12 months back.

I guess why that same sandisk wants to work on the JP console is a mystery.

ramapcsx2, 2023-05-02:
The card is so far removed from the emulation, it is indeed hard to come up with ideas.

L10N37, 2023-05-02:
if it helps, the SanDisk works on all systems when it isn't loaded right up near capacity

ramapcsx2, 2023-05-02:
Something is very wrong here, and I doubt it has anything to do with the console region or the likes.
The green blink is coded to just signify that the SD card could not be initialized. This is a condition where maybe the file system on it is corrupt, or the card slot has bad contacts, those things.
It is fully independent of the console itself, except maybe for power delivery.

L10N37, 2023-05-02:
I was waiting on a mount to put myself together a keeper for once, it arrived today. Times tough with an assignment a week until July/August and customer consoles and orders but  I love my PS1's and always sell the ODE ones. If it reads the card (I know BIOS shouldn't matter , but it will be a JP system) I'll make a video this weekend.

ramapcsx2, 2023-05-02:
You can take the xStation out of the system and power via 3.3V and ground to the UART header... See whether it still hates one card.