x-station/xstation-issues #291: Not really an issue but curious

Enecororo, 2023-05-09:
How difficult would it be to add a feature where you can see the cover art of each game in the loader? (Like some versions of OPL can do for PS2 games)

I can't pretend to know how easy or difficult it would be, but just thought I'd put that out there in case it is possible to add. I feel like it would help me be a lot less indecisive when it comes to picking a game to load

ramapcsx2, 2023-05-09:
Hey,
it's been asked for a lot :)
The basic rundown is: The loader's code base (Unirom) goes open source, then developers colaborate, doing an optional UI part for this, while I add the backend support.
It can be done, it will be slow, but we'll just have to see :)

Enecororo, 2023-05-09:
Thanks for dropping the information. (I admit I only just realized you're the guy that designed the thing lol)

Just hope that UI part will be easy to find and implement if it does become available 

I actually do have another question, would it be possible to program the loader to omit track numbers from the file names?

I find it to be a bit of a blemish on my selection and I obviously can't remove them from the files themselves without compromising their integrity

ramapcsx2, 2024-02-14:
Hm, things take a long time, I got to admit. Unirom open sourcing was going ahead nicely, but I suppose Sickle (maintainer) got sidetracked with other work. It might happen any day though, as it was kind of ready last I checked.
Option loader stuff means better file name cleanup? :)

Enecororo, 2024-02-14:
Fair enough fair enough.

But yeah I'd appreciate a file name cleanup system.

Hopefully it works out

ramapcsx2, 2024-02-14:
There is a system :)
It's balancing processing work and RAM usage with cleaning up file names and removing things like tracks 2+ of multi .bin dumps.
I think what you notice is the display of remaining "Track" mentions in the rest of the file name strings. I can check out whether it can be better though, doing some loader work anyway atm.