x-station/xstation-issues #292: Launch edition of Tomb Raider (USA) takes longer to load than any other game I've tried

Enecororo, 2023-05-11:
not only is it a good 5-10 seconds slower but I also see a "disc tray closed?" notification on the launcher. Generally unsure what's going on there

ramapcsx2, 2023-05-12:
Hey, that sounds like the extra large TOC of the Tomb Raider games is causing issues. 
I don't get that behaviour here.. Could you do me a favor and try using a different SD card, see if that fixes this issue?

Enecororo, 2023-05-12:
Not sure what a TOC is but that is the only SD card I have. My apologies

Enecororo, 2023-05-12:
If I had one I would try it out. I'll just say that it's only an issue with the loader and a minor one at that, just wanted to check in on it

ramapcsx2, 2023-05-13:
Ah that's a bummer. Really any other card would work for testing this, but oh well, I can try to reproduce it .. just need to find a little time :p
The TOC on a CD is the table of contents. It lists all the tracks, and the Tomb Raider games have tons of CD Audio tracks, which could somehow cause this.

Enecororo, 2023-05-13:
It is specifically the launch edition rather than the one with a bunch of demos on it. and it is a sandisk extreme pro so that makes it more confusing lmao

ramapcsx2, 2023-05-14:
Could you tell me which console revision you have? PU-18 or PU-8, cheers :)

Sometimes cards have a few particularly slow sectors in a region, and if the file system information for this game landed on those, it could cause significant slowdowns.
So another test is to simply copy the game again to the card (not deleting the old version).
It will land on different sectors, and if that fixes the problem, well there it is :p

Enecororo, 2023-05-14:
Do you mean PS1 system? Not sure how I can find that out. It is a 5000 series