x-station/xstation-issues #293: Issue with my Memcard Pro (Could be related to the Xstation but I also don't know anywhere else to do this)

Enecororo, 2023-05-13:
Whenever the Memcard Pro is accessed, it tells me the memory card needs to be formatted every time. and I even went as far as reformatting the microSD and installing the firmware again and the issue persists.

Could it be the Xstation that's causing it? Or is the microSD in my Memcard Pro bad?

ramapcsx2, 2023-05-13:
It could be a defect with the MCPro. The xStation is unlikely to cause this.
Best to get support, either on Discord or via mail? :)
https://discord.gg/GyQ8aS9X

Enecororo, 2023-05-13:
Thanks for pointing me in the right direction

closed 2023-05-13 by Enecororo

Enecororo, 2023-05-13:
If anyone runs into the same issue, it turns out the problem was that third party controllers like the Guncon can cause interference with the Memcard Pro's operating system due to data noise from the cable.

8bitmods is working on a possible solution