x-station/xstation-issues #294: (Early pu-8) Xstation doesn't boot Power blue led on / Access green led flash once then continuously for few secondes.

sicounet, 2023-05-17:
Hello , i will try to explain better the title .I am sure the work with the QSB is good . Same for the lifted pins , it's clean and all the corrects pins are lifted .I've triple checked everything.  I did few xstations on pu-18 before with no issue at all . 

When i power on my ps1 after the sony logo i have a long wait that makes believe it will boot correctly to the PS logo and then the xstation... but no . I always end up in the memory card/audio menu . 

The blue led is OK . But for the green access led , it blinks once then it blinks continuously and very fast for few secondes and it stops and the same happen again , one blink then the long flash for few secondes  and stops ...forever ...
I know about the blinking that show that the sd card is not recognize but i can't find anything related with my blinking green led problem.

I use a sandisk 128go , the loader.bin and update.bin are in the 00xstation folder. And i also tried with other sd cards.

Are there compatibility issues with the early pu-8 that i don't know ?

sicounet, 2023-05-17:
I don't have another ps1 for now . Well yes i have but only 100x and only one qsb for this early version. What you want to say is that the pu-8 can't update the firmware by itself ? 

For the CD player , yes it was running games before but i didn't try with a simple audio CD if this is what you meant .

ramapcsx2, 2023-05-17:
Please do the CD player test. It doesn't matter what image type is mounted for this.

sicounet, 2023-05-17:
ok... i can put the cd player again even with the xstation installed ? Because now , with a cd player , the laser go crazy if i let the xstation connected. without the xstation i only have a black screen ....

ramapcsx2, 2023-05-17:
The test is for the xStation, so it has to be installed normally. It is to tell me a bit more about what works and what doesn't.

sicounet, 2023-05-17:
ok , then when i start the ps1 it boots normally , and when i close the lid , the cd starts to turn but the laser immediately go on the outside edge and it makes krrrrrrrrrrrrrrr , so i turn the ps1 off ....

ramapcsx2, 2023-05-17:
There should be no drive, just the xStation...
You're supposed to run the modded console as if everything was working, with xStation and SD card and all :p

sicounet, 2023-05-17:
this with 2 cd player. also , i have a igr installed , i forgot to mention it.

sicounet, 2023-05-17:
haaaaaaa sorry , i understood bad . i thought you asked me to run a disk in the cd player . ok i will try this. sorry

sicounet, 2023-05-17:
ok sorry about that ... so ... it was hard to enter the cd player because when the access light works, i can use the controller. But when it stops i can move for a few secondes until the led starts again. anyway... 

1 track i found and it runs correctly , the timer runs.

ramapcsx2, 2023-05-17:
Okay, this means that most signals work correctly, and it tells me that the xStation should boot the loader on this machine (no update required).

I think the machine can't decode data. There could be an issue with signals 4, 14, 15 here:
[image]

sicounet, 2023-05-17:
For me i am good , i have tested the continuity many times . if i had made bridges under the qsb i would see it with my multimeter too i think . but i think i am ok .

ramapcsx2, 2023-05-17:
You can try a variation of the CD player test.
If you replace /00xstation/loader.bin with an audio CD image (or a game that has CDDA tracks, like Tomb Raider), it should have audio when you play it. The audio should be clear, without obvious distortions.
If you get silence though, then data is not even arriving in the console.

sicounet, 2023-05-17:
i replace loader.bin by a cd image or the iso of tomb raider for exemple ? all this inside the 00xstation folder ?

sicounet, 2023-05-17:
not sure if i am doing it wrong or not . I only have a single track that runs normally but with no sound like the first test. I used a tomb raider game iso . I should see all the tracks of the game separately in the cd player if i was doing it good ?

ramapcsx2, 2023-05-17:
Yep, complete with .cue file and i forgot to mention that the image and cue file need to be named "loader.bin".
The .cue also has to be edited so that the "FILE" entry says "loader.bin", instead of the original game name.
Please use any image that is a single large file, not many small ones, as that is too tricky to rename correctly, etc.

sicounet, 2023-05-17:
... i don't understand sorry . i can't run this 2nd test . But in the end , imagine i managed to do this other test . If i have sound what does it mean , and if i don't have sound what does it mean ?

ramapcsx2, 2023-05-18:
Sorry, this isn't a well planned debugging procedure, so the instructions are vague. Please just simply use the attached image and place it in /00xstation/, overwriting the old loader.bin in it.
Now start the console into the CD player and try to play the single track 1 (should be about 2:30 or so) and report on the audio quality.
[cddaloader.zip](https://github.com/x-station/xstation-issues/files/11505427/cddaloader.zip)

sicounet, 2023-05-18:
ok , thanks , this worked . i have a 2mn30 music and it runs perfectly . Clear sound.

ramapcsx2, 2023-05-18:
Okay, that completes the test cycle we can do here. The system should work.

Since it doesn't, I'm afraid the CD decoder chip might be bad. It is not uncommon for early boards to have defective CD decoders..
There is still a chance that the problem is something weird that isn't covered by the CDDA test, but that is what I think is going on :/

ramapcsx2, 2023-05-18:
I've repaired such console before, but it always involves swapping chips from donor boards.
So yea, maybe best if you take another console.

sicounet, 2023-05-18:
and what chip it is ? swapping it i think i can do . i can use the cd decoder chip from any other ps1 or i need the same model ?

sicounet, 2023-05-18:
ok , it works perfectly with another ps1 . You were right . I thank you a lot for the time you gave to me . 

I am still interested to know which chip the cd decoder chip is. For my knowledge .  

Thanks

ramapcsx2, 2023-05-18:
Look for CXD1199 or CXD1815, but I do not recommend that you blindly swap those chips.
I would normally check a number of signals on the oscilloscope, and try to narrow down which chip may be bad.
Sometimes it's not the 1815Q, but its RAM chip. Other times, it was the SPU.. This is a tricky repair thing.
Good luck :)

closed 2023-05-21 by sicounet