x-station/xstation-issues #296: Audio issues after install.

Kingkongworm, 2023-05-21:
Tested 5501 prior to install. Now almost All none streaming audio is distorted and sounds like a feedback loop…I think I may have accidentally completed a circuit I shouldn’t have somewhere but I just want to know which points are responsible for these sounds. Streaming audio is fine. Boot up sound is really bad.

ramapcsx2, 2023-05-21:
That's not good. There is little relation between the xStation install and the SPU system that is responsible for sound generation.
Please take some pictures of your work, maybe I can spot something.
For reference, find the SPU and its SPU RAM here:
http://wiki.psxdev.ru/index.php/File:PU-18(a)_blocks.jpg

Kingkongworm, 2023-05-21:
Here a pictures of the PU18. I accidentally bridged some connections and pulled up (and soldered back) the wrong joint on the SPU. I tried to carefully get them back into place. 

[image]

[image]

previously I had tried this with a PU8 that had a none functioning disc drive, but it still only boots to the memory card screen so i think there might be something else wrong with it…any thoughts? I have the pictures of that less nicely fluxed and soldered attempt below: 

[image]
[image]

thanks for taking time to help. I’ve always just done little soldering projects over the years and this is proving to be a mite above my skill level and leaving me very very frustrated!

Kingkongworm, 2023-05-22:
Okay! I accidentally lifted a pin on the SPU instead of the CD DSP…let that be a lesson to fellow morons. I soldered that pin back down (luckily I had only ever so slightly lifted it) and it works great. The s video output is not as nice as my other PSX so I might try to remedy that, but otherwise I think I’ve had a successful install after all. Time for Sexy Parodius.

closed 2023-05-22 by Kingkongworm