x-station/xstation-issues #297: ps1 stopped giving video

jonathan3000aya, 2023-05-29:
Any idea what could have happened? After the installation and verifying that the system was working, I proceeded to update it and put games on the SD card, everything was fine. But today, when I tried to start the PS1, the screen blinked and there was no video or audio. I checked the connections, and they are all properly connected.
when turning on the console the blue led stays on and the green led only blinks once with the sd, without the sd the blue light stays on and the green light turns on every 5-6 seconds

ramapcsx2, 2023-05-29:
Please check the APLL wire, make sure that it is free and won't be crushed by the metal shield, which would short out the signal.
It sounds like such an issue :)

jonathan3000aya, 2023-05-29:
I don't see any short circuit, maybe there was, could it have fried the system?

jonathan3000aya, 2023-05-29:
I have tried to turn it on outside the case and it still does not give video

ramapcsx2, 2023-05-31:
Okay, so the LED indicators on the xStation tell me that it is fine and working.
The error needs to be debugged, but besides the APLL wire, I don't have enough information to guide you..

ramapcsx2, 2023-06-07:
Hey, I want to help you fix your console, but what you have is a soldering problem that is not likely to be solved here in an issue tracker on GitHub.
You need to provide pictures, you need to measure test points for bridges / shortcuts, that kind of thing.
If you want to, please head over to Discord, there is an xStation help channel, and I'll try to guide you: https://discord.gg/kMSmCkz3

The other way to fix it: You could send the console to a repair man :)

jonathan3000aya, 2023-06-07:
> Hey, I want to help you fix your console, but what you have is a soldering problem that is not likely to be solved here in an issue tracker on GitHub. You need to provide pictures, you need to [...]

I tested the continuity of each point and they were all connected unfortunately I tried to unsolder the cable and solder it again but the base broke now I'm waiting for the replacement

jonathan3000aya, 2023-06-07:
[image]
[image]
[image]

ramapcsx2, 2023-06-08:
The pictures look good to me. It's good soldering, and the mod seems to have worked for a while, right?
I hope the replacement QSB solves the problem, when you install that :)

jonathan3000aya, 2023-06-08:
Yes, I think I can make a short with the metal base, maybe I'll burn some circuit