x-station/xstation-issues #298: fearful harmony

prank175, 2023-06-09:
Hi, fearful harmony problem, I can't understand why.
the cd player with the lid closed works.

ramapcsx2, 2023-06-10:
Hey,
that's not a lot of information, so I can only give some vague tips..
Since the CD stuff works in the audio player, and since you get some kind of black Sony screen, the xStation is mostly working right.
There could be "almost-shorts" on the QSB, where stuff like too much flux connects 2 data signals. Check and clean if necessary.
The console could also be bad. Did it work with a drive before the mod?

prank175, 2023-06-10:
hi, thanks for the quick reply, I think is a console problem, as you said cd player works and enter in licensed screen, do you know which ps1 chip could be responsible of fearful harmony?

prank175, 2023-06-10:
about the drive unfortunately I can't remember...

ramapcsx2, 2023-06-10:
The problem can be from a chip defect, or from bad data from the drive (xStation in this case), and it's best to first check the QSB in detail, looking for any shorts / "almost shorts".

prank175, 2023-06-10:
I checked the connections, all good, no shorts. Sometime when I power on the ps1 stays still on the first sony screen, some times goes in the bios menu. I tried also freepsxboot, but it blocks on"seeking..."

ramapcsx2, 2023-06-10:
You also need to check and fix any "almost shorts", which is harder to test:
You want to measure signals against each other, in multimeter resistance mode. If any pair shows a very low ohms reading, so maybe below 2000 ohms, it is suspect. Extra flux, dirt, tiny solder bridges can cause these and need to be fixed.