x-station/xstation-issues #299: I installed DFO on top of xStation now my PlayStation SCPH-5502 is not booting (No Signal/Black Screen)

Mo7Jameel, 2023-06-10:
[image]

Mo7Jameel, 2023-06-10:
Note: when powering the PlayStation the power led indicator turns green but no power coming to the controller ports or the MC ports and the no lights indicators on the xStation.

ramapcsx2, 2023-06-10:
Okay, so whatever changed, check where it could have blown an 8V line fuse, because that's (most likely) what has happened.
No 8V at all on the xStation means no power to it, hence no blue LED. Without xStation power, the PSX CPU has no clock signal, hence no picture.
If your TV actually says "no picture", it is possible that more than a fuse fried.
If the TV has sync, but all it shows is a black screen, it probably is just an 8V fuse.

closed 2023-09-21 by mohdjameelbh