x-station/xstation-issues #301: SD Card needs rewrite every X time

NaixZalion, 2023-06-22:
Hi all! 

This issue is extremely strange and particularly. But is not the first time I experience it. 

The facts are. I have an Xstation with Sandisk 512 GB A1 Clas 10 (An SD card with the best quality, buyed new on Amazon by 60€). I installed it on September 2022 with around 300 games. It worked great. Perfect sound, no issues on FMV, quick load... I remember I was playing to PS1 during the Christmas diferent days with no problem, 0 issues. Everything great. After that my PS1 has been stored by 6 month. 

My surprise comes now. June 2023. I take my precious PS1. I select a game and then happened: FMV have stutter, the audio have skips... WTF?? I've test 20 diferent games, and all of them have problems on the videos, music and loading moments. 

After that, I did a copy on my PC with all the games on the SD card, formated the card on exFat (with the program metioned on the XStation guide) and paste all the games on the sd card. Now everything works great again. 

The reason to share that is, that I experienced a similar issue with other ODE one year ago. With GDEmu on Sega Dreamcast with a Samsung SD Card 256GB. Every 6 month the data was corrupted on the SD Card or something, and It need to copy, format sd card and copy again the games to do work the GDEmu (the same as XStation). But... On GDemu I solved the problem buying a Sandisk SD Card. On Xstation, I don't know what I can do to solve it. 

I've many machines with SD Cards: modern consoles, cameras, everdrives... This is the first time I see that the past of time can affect in any way the data inside the SD card. 

- Anyone could explainme why happens this especially with the ODE? 
- Why they are so sensitive with the SD Card? 
- I could do something to be sure that the SD Card will remain correctly with the games that have inside during the time? 

Please, giveme light on this. This issue has made me almost crazy in the past.  

Thank's

ramapcsx2, 2023-06-24:
Hey,
I have no solid answer: It should not happen.
But what might be going on is that the card has weak flash cells that need lots of "reactivation" bit shuffling after a while.
It is a condition in which the card controller moves data around a bit, to refresh the flash charge.
It would be unusual on a Sandisk, but who knows..

Ameno-Sagiri, 2024-11-14:
I wanted to comment here to draw more attention to this issue, just to contribute something to someone looking at this in the future. 

I am in the EXACT same situation as you. I bought a 512GB Sandisk Ultra (A1 Class 10) Red/Silver Micro SD. I purchased a pre-installed XStation PSX in August 2023 from a reputable seller. 

It worked great until sometime around February-April 2024 (about 6-8 months later), where sound files and FMVs started lagging and stuttering. Initially, the problem was light enough that it didn't bother me too much and I put up with it, now, in November 2024, another 7 months later, I pulled out the PS1 to start playing things again, and the stuttering is extremely bad with horrifically loud popping and crackling sounds on the audio 10x louder than everything else coming through the audio stream. It takes what feels like multiple minutes just to get through 10 seconds of an FMV, which has made playing games with unskippable FMVs impossible to play.

Maybe it should be known in the community that this card is not well compatible with the Xstation since I can now corroborate the same exact issue with the same exact card, or maybe it's however they produced these batches of this card in a specific time frame. Might bring some merit to the card having weak flash cells.

ramapcsx2, 2024-11-15:
Hey,
please try one thing with the card, if you can put up with the uncertainty a bit longer.. :p
Please try formatting it using a very large cluster size, say 16384kB (16MB) / exFAT.

I'm currently testing this on a few slow cards I have, and this really helps these out in terms of access times.

But if you want a new card now, I would actually recommend the latest gen Samsung A2 cards (blue ones, "pro" in their name), or the red/gold Sandisks Extreme.

Ameno-Sagiri, 2024-11-15:
> Hey, please try one thing with the card, if you can put up with the uncertainty a bit longer.. :p Please try formatting it using a very large cluster size, say 16384kB (16MB) / exFAT.
>
> I'm currently [...]

I was using the SD Card Formatter tool that I found as a recommended for Xstation (and that is what I used when I originally got the card for use in Xstation), which doesn't seem to have the option for setting that, but I could try it with Rufus or default Windows formatting maybe? The only problem is I won't know if it works until 5+ months down the line as it seems the card just... decays in ability over time until being reformatted.

I have just confirmed after my previous post that formatting the card did fix the issues (for now). I also noticed that certain games that wouldn't load before (like GTA 2), now load fine, so the issue was also preventing certain games from loading which I forgot to mention.

I'll try formatting the card again, and I'll probably use Rufus for it, unless you have another suggestion for a program to use.

ramapcsx2, 2024-11-15:
I used the Windows formatter here. The large cluster size options are further down the dropdown list.

Your card must be particularly bad, to develop so many slow sectors over time, but it seems like Sandisk made their A1 lineup have some really bad apples in it now. And yea, it's an entire lineup of "the same" red/gray A1 cards now. You get a "plus", or some speed rating.. it's wild.