x-station/xstation-issues #302: XStation Firmware + Loader [2.0.2] - 2023-02-20 Legend of Legaia USA freeze

whitezombie2000, 2023-06-28:
I saw this issue posted on Reddit by someone and tested it for myself to see if there was a problem. I just did a few auto attacks and eventually the game froze and the camera kept spinning. 

https://www.reddit.com/r/psx/comments/14fmng7/issue_with_xstation_and_legend_of_legaia/
https://twitter.com/whitezombie2000/status/1673867127617462272

ramapcsx2, 2023-06-28:
Hey,
this is indeed a regression, caused by some bad assumptions in the new 2.0 branch.
I know about it and have already found the problem.
An update to fix this game (as well as the other regression, the DDR dance games) will be out as soon as I can wrap it all up.

It's a bit delayed, due to major SDK changes for next release, that I have to test and verify first.
In the mean time, please revert to 1.6.1 to play these titles. Sorry about that!

ramapcsx2, 2024-04-08:
It's in beta now :)
https://discord.gg/rX9ABMFuqA if you want a peek. Otherwise it won't be long now for the proper update release.