x-station/xstation-issues #303: xstation hdd or SSD? no microsd

sigarette1978, 2023-07-01:
Hi ramapcsx2, I'd like to mount a 2 terabyte type hdd or SSD on the psx xstation, could you install it in the future? Is there any project?

ramapcsx2, 2023-07-01:
Hey,
to start of my argument against HDD/SSD, they typical power consumption of the entire console is about 7W on average.
A spinning disk can easily match that, and for an SSD, it depends on the type and generally the power saving options.
It's safe to say though that such devices are seriously unsuited to the console, and would need special consideration just from the power aspect.

My 2nd point after power is economics: 
A drive controller chip can be hard to find these days, and I can only imagine what it may cost. I think it might be 10x the xStation BOM cost by itself.

3rd point ties with 2nd:
Someone has to develop the hardware / software to talk to these things, too :)

So with those 3 points to begin with, I want to argue that it is a bad idea to pair an HDD/SSD with the xStation.
Of course it could be done, but does it make sense? I wonder :)

ramapcsx2, 2023-07-01:
Next day, another point :)
The price of microSD storage is always getting cheaper (slowly, but surely). This graph shows pricing for the best 1TB card available over time:
[image]

A card like this is not required for xStation. A SanDisk Ultra 1TB does the job well enough, and currently sits at 79.99€.
One could argue that this option is cheaper in total, than an xStation with drive controller and native HDD support.