x-station/xstation-issues #304: Where to order from?

Androxilogin, 2023-07-20:
I bought a few of these from Castlemania. The past two times, they were extremely slow processing orders, shipping my last order out 9 days after it was received and simply ignoring tickets. I've seen many others with the same problem. Now I went to their store to potentially buy another one despite their lack of care for their customers since it was the only place I've found to get them and it isn't even listed anymore?

ramapcsx2, 2023-07-20:
Hey,
please hold the line, Castlemania specifically got singled out recently, and they are a small enterprise that needs to sort it out.
All their xStation orders are processed and shipped though, so if you had anything in flight, it should arrive soon.
In the meantime, the EU partners are available. I think updates on this might be posted via mail or Twitter, once they are known.

https://shop.modinfrance.fr/en/playstation/54-ps1-xstation-optical-discdrive-emulator.html
https://dragonbox.de/de/everdrives-flashcarts/xstation-cd-laufwerk-simulator-ode-psx-pu-18.html

Androxilogin, 2023-07-20:
> Hey, please hold the line, Castlemania specifically got singled out recently, and they are a small enterprise that needs to sort it out. 
> 

I'm not sure how to take that. My second order was in January and they took forever to fulfill it. They still hadn't had whatever they need sorted out by May, still ignored questions and never followed up. I understand running a small business, I run one myself with a ton of products by myself. My main priority, though, is to insure my customers are up to date with everything that is going on with their order. Perhaps I wanted to cancel it since it hadn't even been packaged after a week and they just sent it out discarding everything I needed addressed. My orders have come through after a long time while keeping others waiting. 

I understand this isn't your concern, but they should not just be excused for treating their customers like crap. $110-$115 is a decent amount to just throw to someone who will leave you in the dark and expect the customer to go out of their way externally to find out any information on what might be going on with their order.

I appreciate your links and I'm more than happy to put my money behind someone else. Thanks!