x-station/xstation-issues #305: Insert CD on a Late PU-8 100X and pad with 14 signal

lilo-san, 2023-08-01:
I'm currently getting the Insert CD screen.

I have measured continuity as suggested on the xtation issues page, everything seems fine.

The only meaningful thing is that when removing the 3 resistors I scuffed the pad used to measure signal 14. 

Is this pad carrying the signal somewhere? I could run a wire if needed.

ramapcsx2, 2023-08-01:
Hey,
the burnt pad should be no problem, it doesn't connect anything anymore.

Your error is unfortunately the generic "something doesn't work" problem, that just means we'll have to try and debug it.
Please do the CD player test first:
Start the console with the lid open, enter the BIOS CD player, then close the lid.
A CD should get detected, and when you press Start, it should start to play silently, with the seconds counter advancing.

lilo-san, 2023-08-01:
A CD gets detected.
When I press Start it doesn't seem to play. The seconds counter doesn't advance.

ramapcsx2, 2023-08-02:
Okay, that's curious, actually.
Let's see... SUBQ bus must work for the TOC to parse like it does... it could be the limit switch signal, maybe.
Check your QSB around point "20" here:
https://github.com/x-station/xstation-issues#late-pu-8--21-and-later

lilo-san, 2023-08-02:
I have reflow the points around "20" 

I have go though the diagram measuring for continuity in all points. Check for bridges, make all solder points pretty. Etc...

Same result.

lilo-san, 2023-08-02:
For the sake of completeness allow me to share:

My work station, I use my phone as microscope that gives me the resolution you se on this photos: https://ibb.co/DMQgnCM

Step 2-CXD2510Q-lift-pins-44-63 : https://ibb.co/h77QPnk
Step 2-CXD2510Q-lift-pins-66-69: https://ibb.co/87SdSZP
Step 6-desoldered-filter-and-resistor https://ibb.co/xqpV4yK
Step 8-QSBhttps://ibb.co/k3Kggj3
Step 9-IO16 https://ibb.co/zbq6Rzx

Thank you so much for the time you dedicate you help me.

ramapcsx2, 2023-08-02:
Hm, is there a lot of extra flux in this area? Any excess flux is partially conducting, and can cause signals to not work:
[image]

ramapcsx2, 2023-08-02:
As a heads up:
Your problem must be with a single signal, by the way the CD player behaves. Find that problem, and it should start working :)

lilo-san, 2023-08-02:
So, I once again I did go over the whole QSB and reflow all connection that didn't look like a shinny ball.

It started playing!!! It reached second 28!!!

What does that mean?

ramapcsx2, 2023-08-03:
Awesome, well, does it now boot the loader as it should? I suppose not, or else you would have mentioned? :p 
But yea, the reflow fixing this shows where the problem was. If it persists, try to check the area more.. something still off there.

lilo-san, 2023-08-03:
No menu yet. Green led  blinks twice.

I guess you suspect the same area again?

ramapcsx2, 2023-08-03:
Yea. It's just safe to assume that if there was an error once, there can be more of the same kind.
It's hard from here on to give more tips. Whatever you can provide may help :)

lilo-san, 2023-08-03:
Finally reach the menu!!!

Do we have any kind of test suite to ensure everything works correctly from now on? Or some games that put to test key things?

Thanks for all the support

ramapcsx2, 2023-08-04:
You're welcome, glad you fixed it :)
There's no special test disk, as all you really need is to run a couple games (or just one, really) to tell :p 

I'm on Discord every day, have it open while working:
https://discord.gg/M4DYXTpp

lilo-san, 2023-08-04:
Reflow 3 more points. Cold Boots to menu now :)

ramapcsx2, 2023-08-04:
Oh right, that's also an option in the the loader, so when the hardware all works, you can pick the cold boot behaviour there :)