x-station/xstation-issues #306: Freezing on Loading Screens in Dance Dance Revolution Series

ronyket, 2023-08-03:
Got finally my XStation installed on the following setup:
2.0.2 Firmware + Loader
SCPH-1001
PU-8 Board
128 GB SanDisk Ultra MicroSD Card, formatted as FAT32*
All ISO files from Redump collection, each game in separate folders, separated by genres.

*I couldn't get the loader to recognize all games when formatted in ExFAT, using the Tuxera SD Card Formatter, the only format that the loader recognized all the games on the SD Card was FAT32 (formatted with 32KB clusters via guiformat).

Most of the games are running great, but some of the games from the Dance Dance Revolution Series, freezes on loading screens, when accessing the Options menu.
The games that hangs up on that screen were:
DDR 3rd MIX
DDR 4th MIX
DDR 5th MIX
DDR Extra MIX

Thanks!

ramapcsx2, 2023-08-04:
Yes, I know. I'm so sorry! ><
It's a problem with the way I implemented the new no SD card mode. These games do something special with the boot sectors, and they detect the change.
I have it on the fix agenda :)

xRoseQuartzx, 2024-03-19:
Hello!

I ran through all of the above games as well in both the USA and Japan versions and noticed the same thing.  Has there been any solution?

ramapcsx2, 2024-04-03:
Putting a note here that these should be fixed upcoming release.