x-station/xstation-issues #307: Pu-7 insert playstation cd

Rexarrrr, 2023-08-10:
I installed xstation on a PU-7 board without using QSB. When loading, the message "Insert playstation cd" Japanese. Point 3 has been checked. What could be the problem? Maybe you need to initially update the firmware version to the latest on another playstation? Was pu-7 support announced only in the latest firmware?

ramapcsx2, 2023-08-10:
You definitely need to be on a recent firmware, but I can't quite remember where I added which PU-7 fixes. Best go to 2.0.2 directly.

Rexarrrr, 2023-08-10:
How can I update the firmware without buying a second playstation?

ramapcsx2, 2023-08-10:
Oh gee, PU-7 (as well as EARLY PU-8) were later additions. You're not meant to drop a release xStation into one of them right away.
It will take time until newly shipped xStations come with compatible code.
For now, is buying a whole extra console really the only option you have?

Rexarrrr, 2023-08-10:
Updated firmware to another playstation. Now everything works great on PU-7. Thanks

closed 2023-08-10 by Rexarrrr

ramapcsx2, 2023-08-10:
Cheers, and thanks for bringing it up. Gonna have to think about it :)