x-station/xstation-issues #309: 8Bitdo PS receiver and memory card not working together

jcpurdy00, 2023-08-16:
I got my hands on the new 8BitDo PS Retro Receiver.
Works great on my PS2, so I know the receiver is good, but it stops my Xstation equipped PS1 from reading memory cards while using it. 

Was wondering if it's a power draw issue perhaps. I don't really know. 

Anybody else have this problem?

ramapcsx2, 2023-08-16:
It sounds similar to the troubles Blueretro has on some PSX consoles, which sometimes have an xStation installed.
There is a massively long issue for it over there:
https://github.com/darthcloud/BlueRetro/issues/102
It's still pretty much unsolved though. We have ideas, but nothing is conclusive enough.

jcpurdy00, 2023-08-16:
I appreciate the response and the reference. 
I'll update if I find anything.

ramapcsx2, 2023-09-06:
I've got an update, kind of. It seems the new 8Bitdo receiver is not behaving well, causing issues even with official Memory Cards.
This is from what 8bitmods investigations found... Best check for firmware updates for the receiver.

FloppySoft, 2023-09-06:
8BitDo fixed it in Firmware v1.03 of the Retro Receiver for PS. You'd need to update the receiver.

jcpurdy00, 2023-09-06:
Thanks for the update. 

Unfortunately the new firmware for the receiver has not fixed the issue for me.