x-station/xstation-issues #311: Xstation install on a 5501 - PU18 QSB

bearwrestler, 2023-08-20:
Hey all, performing this mod on a PS1 5501 that couldnt read disks but powered on. Performed install and get to the CD Player/Mem card screen. Blue light on, two green blinks on start. 3rd party shell. Ive:

- verified pins are lifted on correct chip and no pads are bridged nor legs crossed
[image]

- tested continuity on all QSB points successfully

Any help would be appreciated!
[image]

ramapcsx2, 2023-08-20:
Hey,
CD Player test time then :)
Start the console with the lid open, enter the BIOS CD player, then close the lid.
A CD should get detected, and when you press Start, it should start to play silently, with the seconds counter advancing.

bearwrestler, 2023-08-20:
Hey Rama! Interesting results - with shell on, no CD detected. When pushing down the sensor, it freezes then switches to an insert CD menu.
[image]
https://photos.app.goo.gl/wfZbpW2xddtBaPjU6 for video

ramapcsx2, 2023-08-21:
So the CD player doesn't see any disk either?
This is a multiple failure thing then, though one root problem could cause both symptoms.
For now, please try to reflow every QSB solder point. I suspect at least one connection isn't good.

bearwrestler, 2023-08-21:
Would you look at that!
[image]

bearwrestler, 2023-08-21:
out of curiosity, can you tell me about the function the cd player test is supposed to perform?

ramapcsx2, 2023-08-21:
Nice :)
The CD player uses only basic functions, compared to starting a game.
For just a CD to play, things like the license symbols and even all sector data can be missing or bad.
If a console can do the CD test, but doesn't boot games, I would know that the problem is with data or the license, for example :)

bearwrestler, 2023-08-21:
Thanks for the explanation! I think we're set here for this issue. Thank you for your help 😁

closed 2023-08-21 by bearwrestler