x-station/xstation-issues #313: Space Debris (Europe) doesn't start on quick or full load

alebagran, 2023-08-31:


ramapcsx2, 2023-09-04:
Hey, just got back. I'm looking at it asap :)
The .cue file I got was a bit strange looking. It might just be that..

ramapcsx2, 2023-09-05:
Oki, pretty sure this game is very broken. Be it the dump or the actual game, it's pretty bad.
Check your .cue file whether it says (EU) or (Europe) in the file name, btw. Mine was mismatched, saying (EU) wrongly (not matching the .bins).

ramapcsx2, 2023-09-05:
Oh, now I realize why it's so weird here (testing using a different ODE). This is also libcrypt enabled. Xstation should deal with that fine though.

alebagran, 2023-09-05:
My cue file says (Europe) in the filename as well as the bin ones. The redump one I got from here: https://archive.org/download/redump.psx.p3/Space%20Debris%20%28Europe%29.zip

ramapcsx2, 2023-09-06:
So the game runs normally here. It is fully listed in all variations for libcrypt, so I don't know what else could be wrong for you ><

lamprey13, 2023-09-14:
Do you have an NTSC PS1? I forget the different versions and there limitations, but I have a 1001 and I needed to do a PAL to NTSC "patch" in order for any Europe/PAL games to work on mine. So, yout might look into that if you are having issues with PAL games. You can do a search for a tool to do it, as there are several. I used Zapper 2020.

alebagran, 2023-09-26:
Thanks @lamprey13. I'm going to try it