x-station/xstation-issues #314: Certain games won't start if written on SD (exFAT) from macOS 10.15

nehalem501, 2023-09-03:
I found a rather strange bug, and it took some time to find the source.
After copying new games to the microSD card and refreshing the game list, certain games wouldn't start (when selecting the game, I would get the PS1 boot logo and then the CD player).
I checked if the files were badly copied (did checksums of the files on the card and everything from different OSes) but nothing seemed off.
But then I noticed that if I deleted these games, and copied them again to the microSD from another OS such as Linux or Windows (with exactly the same file and folder names), everything worked properly (I'm talking about the exact same games, with the same checksums on the disc images).

The microSD (512GB SanDisk) was formatted as exFAT, the macOS version was 10.15 Catalina and latest firmware on the x-station (2023-02-20).

nehalem501, 2023-09-03:
If you can't reproduce the issue on your side, I can try to reproduce it and send you a disk image.

ramapcsx2, 2023-09-04:
Hey ho,
I've since forgotten again what the other Mac file system stuff was, but if you have another OS to check, could you check out whether any metadata files and folders are nearby the non-working games, please?

nehalem501, 2023-09-13:
Yes when I had these problems I checked that all the macOS dot files were removed. The X-Station had no trouble at all to detect the games, it's only running the games that wouldn't work.

ramapcsx2, 2023-09-13:
Could you drop by in the Discord channel please? (If you can make the time :) )
https://discord.gg/fNMefuJC

nehalem501, 2023-09-13:
Yes, I've joined the server, you can ping me there.

ramapcsx2, 2023-09-27:
@nehalem501 and me had a chat about it, and once he's back home, he'll try to create a low level copy of his card for me.
I assume there are metadata files that break my not-so-great file walker. 
It'll eventually get fixed :)

Darkmelman, 2023-09-27:
It looks like i have found the reason under linux: fuse

Since Kernel 5.7, fuse and exfat-utils are not longer necessary. I only installed exfatprogs and now, the mount command returns type exfat instead of type fuseblk.

Now copied files are running fine!
Hopefully its not a coincidence

Darkmelman, 2023-09-28:
> It looks like i have found the reason under linux: fuse
> 
> Since Kernel 5.7, fuse and exfat-utils are not longer necessary. I only installed exfatprogs and now, the mount command returns type exfat instead of type fuseblk.
> 
> Now copied files are running fine!
> Hopefully its not a coincidence 

Sorry!Wrong all clear!
The problem still exists.

UrbisR, 2025-07-06:
I may be having a similar issue, having formatted the card under Windows 10 but then using Ubuntu to transfer the files over (I unzipped them directly to the card using 7z command line to save space on my PC).
Most games work but 'Vagrant Story' and 'Vandal Hearts' for example are two that do not, and boot into the music player. It's possible that the error happens more with games later in the alphabetical order, as I transferred the games over in alphabetical order and the card may have been filling up. I am imagining disc fragmentation or something?
I also double checked a few of the problematic .bin files against redump and they matched, so I don't believe it is a problem with the image itself.

UrbisR, 2025-07-08:
I decided to go ahead and back up all the files from the SD card onto a Windows 10 laptop, then reformatted the card with the same official SD card formatter again, and then moved the files back to the newly formatted card. Now, it seems like the games load correctly. At least, the games that were giving problems before seem to load and so did a few others and I didn't encounter any other issues.
What a weird quirk though, I never would have thought that moving files with a different OS would make a difference like this...

ramapcsx2, 2025-07-08:
I suspect there is an issue with the card @UrbisR. If the issue returns after a while, you can be pretty sure that it's about the card.