x-station/xstation-issues #316: Console not booting with parallel port gameshark and X-Station installed

bumpnthump05, 2023-09-18:
Would anybody have any info on this? I have a SCPH-5501 with an X-Station installed. Everything works as it should when no parallel port unit is plugged in, but when I plug either of my gameshark pro parallel port cheat modules into the console, i cannot even get past the PS1 bootup screen. It usually hangs on the black logo PS1 screen. I know both of my gameshark pro's work perfectly on any other console. I ordered an Xploder to see if that makes a difference, but ive seen people say their gamesharks work fine.

ramapcsx2, 2023-09-18:
There should be no relation between EXP port devices and xStation, and I'm running such a setup all the time.
There is probably another issue, maybe with the software on the Gameshark itself.
I know that Exploder Pro for example has code that actively interferes with the system all the time (used to allow you to press a button combination and get into the software while ingame).
If your device has such a feature, try disabling it.

bumpnthump05, 2023-09-18:
Thanks rama, Ill give it a shot, but ive been lost on this. Going to also try to clean the parallel port as well maybe? If not im transferring the X-Station to another PS1 because i really wanna use my Gameshark Pro, I just didnt know if there was something different about the pro that didnt allow it to work properly since it has the "virtual memory cards" built into it.

bumpnthump05, 2023-09-19:
Cleaned the parallel port, cleaned the gamesharks, Still couldn't get anything. Went from locking up like normal to not even recognising its plugged in and going straight to the x-station. Full boot nor fast boot take me to the gameshark. Tried a PS1 that didnt have the xstation installed, both worked flawless. [20230918_192308](https://github.com/x-station/xstation-issues/assets/43628253/c7232548-6242-476e-94fd-84120e826ef5)

ramapcsx2, 2023-09-19:
Hm, I guess I'd have to see whether those Gamesharks do anything out of the ordinary.
To me it looks like they want to patch the kernel somehow. It's hard to explain otherwise..
Are these Gameshark Pro regular "Datel" style carts (if you know)?

bumpnthump05, 2023-09-19:
Its the Gameshark Pro, I did go through the menu of the gameshark and outside of normal gameshark functions I do know they have up to 10 virtual memory cards you can enable on it, one at a time. Its the one on the right, not the regular model on the left. I just bought another SCPH-5501 on ebay last night with a bad disc drive to see if maybe swapping the x-station over to a different motherboard will change the outcome, or if I just need to use a regular gameshark due to the virtual memory cards
[image]

ramapcsx2, 2023-09-19:
Please test this for me:
Place a single .bin game into your /00xstation directory, and name the game "loader.bin".
Now xStation will mount this, thinking it's the loader, and I have a hunch that your game might boot.

bumpnthump05, 2023-09-20:
Hey Rama, i couldnt get it ti lock up last night but its still acting like its not detecting my gameshark. When doing as you instructed, it booted to the xstation menu. I ordered another SCPH-5501 to transfer my XStation into. Im thinking there might be an issue with this parallel port. Ill update this when i swap consoles. 
[image]

ramapcsx2, 2023-09-24:
Hey, as an update, someone managed to test their GS3 Pro device on an xStation console, and it worked there.
So it is probably really some kind of problem with your hardware... I assume at least :p

bumpnthump05, 2023-09-24:
Yeah i got in both my xploder and replacement scph-5501 in today. im gonna do some testing and swap my xstation into the other console tomorrow. Ill follow up with my results.

bumpnthump05, 2023-09-24:
Well, trying to uninstall my xstation 550x qsb, i managed to mess it up. Do you where i can get a couple of replacements in the USA?

bumpnthump05, 2023-09-26:
Thanks Robert!!! 
[image]

ramapcsx2, 2023-09-26:
Awesome, seems it's working well. So far I'm impressed :)

bumpnthump05, 2023-10-06:
Hey robert, just wanted to give you an update. Got my new QSB in, installed it into the other 5501, and everything is working as it should! Gameshark Pro is working perfect as well as the X-Station. It had to be something wrong in the parallel port of the first ps1 I worked. Good news is as well, my PixelFX RetroGEM Shiny Edition is coming in tomorrow to put the frosting on the cake! Thanks again for all the time you spent helping me try to figure out the issue!

ramapcsx2, 2023-10-07:
Oh that's good, it wasn't any problem with the loader then :)
Good luck then with the HDMI mod install, and enjoy! :)

kalm-traveler, 2024-08-03:
I know this is 10 months old but just made a github account to add that I ran into a similar situation and it was 100% the console port's pins being dirty. 

I have 2 GameShark Pro carts, they both tested working on 3 other PU-18 boards, would either do nothing, or freeze on bootup on the one I happened to chose to mod. 
Intermittently would work immediately after dumping 99% IPA in the port connector and scrubbing with a toothbrush. Found 3 solder joints on the motherboard pins (underside of board) that looked cold, desoldered the entire connector and resoldered each pin after cleaning. No change.
Bought some electronics cleaner from an auto parts shop - sprayed that all in the connector and scrubbed with a wire brush.
Bam, consistently working now.

ramapcsx2, 2024-08-03:
... For this day. Check back tomorrow and sob :p

kalm-traveler, 2024-08-03:
oh gosh if it gives me any more trouble I'm desoldering the connector and replacing it with one from one of those other boards that never had issues.

ramapcsx2, 2024-08-03:
Heh, oh well, it's just my experience with these.
Quite often, it's the console connector and the cart connector both that have some oxide on them.
When only contact cleaner works, it might be down to the copper already, and then it quickly deteriorates again.

But hey, sometimes it works :)

kalm-traveler, 2024-08-03:
If it's down to the copper, would it help to apply some Deoxit to the pins? or more crazy, how about doing a little nickel electroplating? Was thinking you could probably use a little piece of aluminum foil, push the mainboard side of the connector pins through it to contact them all at once for the other side of the circuit through the chemical bath.

ramapcsx2, 2024-08-03:
Out of my field :p
I don't think anything but really professional treatment ( material science included ) will do.

ultramecha, 2025-10-22:
I also has an issue where my GameShark wasn't working with my parallel port. After cleaning with 99% IPA using a toothbrush and also scrubbing the GameShark device as well it did indeed fix the problem. So it does work perfectly fine with the xStation, just make sure your port is clean! Oh and make sure the GameShark switch is on of course :)