x-station/xstation-issues #317: Reset won't return to loader, full power cycle required

veogehog, 2023-09-22:
Installed the xstation, all is well, my issue is with the reset though.

I have set the menu option to run the loader at power, which it does fine, when I start a game, everything is good. The issue is when I want to return back to the xstation menu (loader screen). If I use the reset button, it just resets the same game I was playing.

I hold the reset button for 1 second/5 seconds, it still just resets the game I'm playing.

I have the x-reset installed too, and if I press L2+R2+select+X it still just resets the game and won't go to the xstation menu (loader screen).

The only way to get back to the  xstation menu (loader screen) is to power cycle the console.

Any ideas what could be causing this?

ramapcsx2, 2023-09-23:
Hey,
could you try "full booting" a game?
This will cause the xStation to pull the reset line, and if that doesn't work right, we know that the fault is with that reset line not reaching the xStation (or similar fault).

veogehog, 2023-09-23:
> Hey, could you try "full booting" a game? This will cause the xStation to pull the reset line, and if that doesn't work right, we know that the fault is with that reset line not reaching the xStation (or similar fault).

So when I "full boot", the game hangs. e.g.

"Bloody Roar Track 1.bin
Loaded System.cnf from Sector 0x13A

then it hangs.

ramapcsx2, 2023-09-23:
Oki, it means the reset signal is not reaching the xStation, most likely from a cold solder joint / no connection.
Please check signal #17 (the reset line) shown here:
https://github.com/x-station/xstation-issues#pu-18

You can check against an alternative point that's also marked:
https://github.com/x-station/xstation-issues#xstation-cant-reset-the-console-full-boot-not-working-after-updates-no-automatic-reset

veogehog, 2023-09-23:
It's a clean connection at point 17,

Using a multimeter also shows the points are connected all ok.

What i might do regardless is put another wire to the alternative point up in the top left corner. See if that helps.

veogehog, 2023-09-23:
I ended up wiring a new wire along point 17 from the QSB to the pad on the mainboard.

I also just touched up the connections.

That's worked.

Not sure how because all my connections were OK (at least when tested with the multimeter), so I can't say for certain whether it was the extra cable or reflowing the solder which fixed it.

But thank you, this is great now.

ramapcsx2, 2023-09-23:
Good to hear, enjoy! :)

One reason a multimeter beep test can be wrong is by the pressure of the probes on the solder connection.
While trying to probe, you can close the circuit just by pressure.

veogehog, 2023-09-24:
ah that's a great point i hadn't considered that!

veogehog, 2023-09-24:
As ramapcsx2 said, it was most likely the solder joint not being perfect, but i was closing it with pressure from the multimeter probe. 

Re-doing the solder joints and adding a second wire as per the guide for signal 17 worked.

closed 2023-09-24 by veogehog