x-station/xstation-issues #319: Stuck on "working"

mlalgjar, 2023-09-28:
I've got a 5501 with no other mods installed other than xstation. I got it to boot into an older version of xstation because the latest released would not work. 

A couple of things I've noticed was that during the PlayStation logo boot instead of licensed by it says "ed by ed by"

When it gets past that and loads into loader.bin, it says "working" with the accompanying animation but does not get past that point. I tried three different micro SD cards one SanDisk one lexar and one generic all with the same result. I checked the SD card and the only thing that gets written to it is a config.txt with nothing in it.

I'm currently looking for a way to purchase the 5501 qsb since I've already damaged some traces on the original from moving it from one PlayStation to another. Continuity still checks out between all the test points and pins on the qsb so I don't think it's that.
[image]

I took the picture before I taped down the CD lid sensor

ramapcsx2, 2023-09-28:
Hey,
this most likely means that one of the signals xStation needs is only partially connected, and it drops out all the time.
It is connected enough so that the loader can boot at all, but it then gets stuck waiting for some response here.
I recommend a careful inspection of the solder work with some retouching.

mlalgjar, 2023-09-28:
I might have to use the slow solder board that I got with the Pico station. Even with my continuity test my qsb is damaged in some places. Since I can't find it anywhere is there a map for a slow solder board for a 5501?