x-station/xstation-issues #320: When using picopsu, full load will freeze

SoukinOKATU, 2023-09-28:
Due to living in a 220V area, I have replaced my PSU on PS1 with an open source PSU called "PS1 picopsu"（Purchase from merchants），It also supports the "soft reset" function, simply connect the handle and motherboard cable on the PSU, as follows：
[image]

But when I use "full load", after loading the image, PS1 does not restart for boot, but directly freezes and crashes

Even if I unplug the 'soft reset' and restore the original cable connection, it will still cause the full load to freeze and crash

It is certain that using this picopsu will cause xstation to be unable to restart PS1. Is there a solution to this? Because this type of PSU is indeed very useful (because it uses a 12V power supply connection)

ramapcsx2, 2023-10-02:
Hey,
I don't really know what to recommend here. The PSU is a design usually associated with Will's Console Modifications, but I don't recognize it as one of them. I suspect it is a reproduction... In any case, a PSX PSU has to provide a certain reset circuit that often gets overlooked by 3rd party PSU designs.
If this circuit is missing, or not standard, it can easily lead to the problems you report.
However, another rather common problem is that the reset signal is simply not connected to the xStation.
Please check point 17 in this diagram:
https://github.com/x-station/xstation-issues#pu-18

nicolasnoble, 2023-10-02:
If this PSU is open source, where is the source so we can verify the presence of the reset circuit, and suggest something if it's not present?

SoukinOKATU, 2023-10-02:
Thank you for replying to me. The picopsu I am using seems to be this:

https://www.willsconsolemodifications.co.uk/index.php?route=product/product&path=59_67&product_id=111

But my version is "ps1picopsu v2" (which seems to be an early version), and it seems to be a clone version. The author said that there may be an error in the reset logic of the clone version, and I suspect that this is the reason for the freeze

nicolasnoble, 2023-10-02:
Right, so it's not open source. It's just a clone, and we can only judge from the pictures you have. It's technically easy to add a reset monitoring chip to these, but it requires some soldering. Not sure if it's worth it, or if it'd be better just to get the real thing from Will.

SoukinOKATU, 2023-10-02:
At present, I have replaced the original PSU (220v) with a PAL version (5002). I will use it this way for now and consider buying the latest version from Will's store in the future

ramapcsx2, 2023-10-02:
This is a good solution. I really did not plan for cases where the PSU does something way out of order (like missing the reset circuit)..