x-station/xstation-issues #321: Black screen on start up

SpikeEscape, 2023-10-02:
First time install on a PU 18, been troubleshooting for a couple days and I can't find the cause.
The console ran normally before the installation, the leds on the xstation appear to be functioning properly, 2 quick green blinks and a solid blue led.

I've confirmed good connections on all the pinouts as guided by the graphic on mmmonkey.co.uk, also tested against ground and there is no accidental grounding on the QSB, just the 3 intended ground points as indicated by the graphic.

I've also checked continuity on the ribbon cable when seated and can confirm that the connections between the qsb and the xstation board are solid.

Using a magnifying glass the lifted pins appear to have no bridging, my last attempt at a fix was cleaning the pins with a q-tip to ensure that there is no flux paste leftover to act as a bridge, and still no boot.

When the console powers on you can see a brief flash on the tv as you normally would when turning on these older consoles, but it is followed by nothing other than black.

Pictures below:

[image]

[image]

[image]
[image]

Note in these pics you can see some fluffs/strings from the q-tip I used for cleaning off paste, I removed this before testing the console.
[image]
[image]

ramapcsx2, 2023-10-02:
Hey,
a total black screen means (together with what you describe) means that the PSX CPU is not getting the required APLL clock signal.
Could you please first start by replacing this wire? I see there's a crunch mark in it. Make sure the wire can't be squeezed between the mainboard and the metal shield.

SpikeEscape, 2023-10-02:
> Hey, a total black screen (together with what you describe) means that the PSX CPU is not getting the required APLL clock signal. Could you please first start by replacing this wire? I see there's a crunch mark in it. Make sure the wire can't be squeezed between the mainboard and the metal shield.

Hey there,

I've replaced the wire with a fresh one and the issue remains unfortunately.

Continuity test confirms that it's solid from the AAPL pin of the ribbon socket all the way to the pad just in front of the wire on the motherboard.

[image]

ramapcsx2, 2023-10-02:
Okay, next, you say that the xStation does some green blink action at bootup, and the blue LED is solid, right?
It means that the main program is running, and the APLL unit should be sending the 67mhz APLL signal to the CPU.

So assuming that is all fine, what else can leave the system in permanent reset?
Is your PSU modifed in some way? Do you have any other mods installed?

SpikeEscape, 2023-10-02:
Yes there's a constant slow green blink with no SD card inserted.

The only thing I can think of is when I was installing I accidentally lifted two pins on the IC308 instead of the IC701, but I was able to tack them back down and clean them to where they looked fine, so I wouldn't imagine that would cause this.

[image]

ramapcsx2, 2023-10-02:
Please make triple sure that those pins are securely reattached. Sometimes they seem solid but have no electrical contact at all.

As a further test, if nothing seems to help, you can try removing the APLL wire, and reconnecting the original oscillator (if you still have the removed resistor, putting that back would be best.. but a simple link wire or solder bridge will also do.).
This test is meant to completely remove the APLL signal and instead use the original clock source again.
If this system then boots to the Sony screen normally, we know for sure that there is an issue with the APLL signal.

SpikeEscape, 2023-10-02:
I was able to follow the traces out of the pins on the 308 and using the multimeter they seem to be electrically sound and unbridged as well.

I've removed the wire and solder bridged the two pads where the resistor once was, multimeter said it was good but yet there is still the black screen issue, after that I tried using different video cables and AC cables but they did not help.

ramapcsx2, 2023-10-03:
Time for a fuse check! :)
It could be that your 3.5V PSX supply line is broken at the fuse.

ramapcsx2, 2023-10-03:
If it says "PSOne", it should be fine. I always use a donor board here :)
But yeah, that may be it :)

ramapcsx2, 2023-10-04:
One note of caution:
There was a reason that the system supply fuse blew. You should try to look for any possible shorts on the supply line that this fuse used to connect to.

rablador, 2023-11-02:
Just wanted to chime in and say that I have the EXACT same problem as OP. Same behavior down to every detail. And I have tried most of the same things as well. ^^; Was there any resolution to this in the end?

ramapcsx2, 2023-11-03:
@rablador So your fuse blew and all that?

rablador, 2023-11-03:
Oh, sorry, no blown fuses as far as I can tell. The 15, 50 as well as L106 and L105 all work. I have done the following:

1. Checked and double checked IC legs.
2. Checked and double checked continuity between PS1 main board and QSB.
3. Randomly checked a few ribbon cable connections, some even down to PS1 main board. 
4. Checked APLL wire.
5. Tried different TV:s and AV cables.

I can troubleshoot with power on if that helps, if I can get some pointers to what to test.

Also, there are no other mods installed.
[image]

rablador, 2023-11-03:
Oooooh, I just saw on my own screenshot that I have connected APLL to the wrong spot! I'll see myself out... 🙄

SpikeEscape, 2023-11-03:
Hey, I meant to come back to this, what I ultimately ended up doing was getting another 5501 and transplanted the QSB to the new motherboard and redid the install successfully, so unfortunately I never found out what caused the fuse to blow or if there were other issues with it.

closed 2023-11-03 by SpikeEscape

ramapcsx2, 2023-11-03:
> Oooooh, I just saw on my own screenshot that I have connected APLL to the wrong point! I'll see myself out... 🙄

Welp :D
If the problem continues, you can start a new issue. We'll go from there.
But otherwise, both of you enjoy :)

rablador, 2023-11-03:
I resoldered the APLL line - works perfectly 😅