x-station/xstation-issues #322: Weird logo glitch then black screen

yesmorepain, 2023-10-08:
Hello

I installed the xStation with the 55xx QSB and I get blue light as well as the bootup green light but when I turn the system on, I get an odd BIOS glitch then it black screens

Will reflow the points just to make sure, but I am confused considering I haven’t seen anyone else deal with this.

[image]

ramapcsx2, 2023-10-08:
Hey,
that is a typical sign of bad CD data bits, so a reflow is a good idea.
Data happens around the point marked *14, as well as the single *15:
https://github.com/x-station/xstation-issues#pu-18

yesmorepain, 2023-10-08:
> Hey, that is a typical sign of bad CD data bits, so a reflow is a good idea. Data happens around the point marked *14, as well as the single *15: https://github.com/x-station/xstation-issues#pu-18

thank you for the reply

i reflowed all the points before i read this comment but im still having this issue

maybe it is an issue with the throughhole

ramapcsx2, 2023-10-08:
Of course, I'm guest on the Classic Gaming Discord:
https://discord.gg/paQBTcdn

rablador, 2024-01-31:
I'm having this exact problem too (same glictched logo). I have reflowed all points, doubly on the points for 14 and 15. I have also checked continuity as per the photo of PU-18 here: https://github.com/x-station/xstation-issues#pu-18.

ramapcsx2, 2024-02-01:
We just had someone on Discord where it was a bad Decoder chip (the 1815Q). Sometimes it's just chip damage :/

rablador, 2024-02-01:
Oh, that would be a little too hard for me to replace. :/ Oh well, it was a rescued machine and a spare QSB anyway. Thanks for the quick reply 👍

ramapcsx2, 2024-02-01:
It's always worth trying to debug it more, but I can't know whether people have the time and tools for this.
Usually with enough effort, it can be fixed, whatever it is. But if you've got a damaged Decoder, it's going to take a replacement chip.