x-station/xstation-issues #323: Short on Pin 8

GenuineHili, 2023-10-08:
Hi, I installed the xStation in a SCPH-5502 today and I only get the memory card/audio player screen. I then measured all the test points on the QSB and found out that I have a short to ground on pin 8. What can cause that?

ramapcsx2, 2023-10-08:
Hey ho,
first of all, where did you find a Pin 8? :D

GenuineHili, 2023-10-08:
I mean the pin 8 on the flex cable connector. But I know now why it was shorted to ground. I bridged the lid switch so I didn't have to press it when I was troubleshooting and the solder point of pin 8 leads to the lid switch, so that's normal I guess.
When I turn on the console, the blue power led lights up and the green led blinks once. Then I end up in the memory card/music player screen.

ramapcsx2, 2023-10-08:
Okay, so this is now a regular "boots to BIOS screen" issue :)

ramapcsx2, 2023-10-09:
Please do the following CD player test described here first, see whether you have a CD playing silently normally:
https://github.com/x-station/xstation-issues/issues/288#issuecomment-1521545393

GenuineHili, 2023-10-09:
Ok I tested the CD player and it doesn't work. I can't press the play button at all.

ramapcsx2, 2023-10-09:
Okay, it means there is a general soldering (signal not connected) issue. Can you take some detailed pictures of your work?

GenuineHili, 2023-10-09:
[image]
[image]
[image]

ramapcsx2, 2023-10-10:
Okay, now I need to guess a ton:
- If you can't press any buttons in the CD player, it might indicate that the system is repeatedly trying to init the drive. This can happen if signal *20 is not connected:
[image]

It can also mean that the whole bunch of these signals has problems:
[image]

Please try to verify that *20 reaches the xStation flat flex connector pin 20, as well as one pin right next to these on the square "mechacon" IC:
[image]

- while at it, sometimes signal *19 is not connected to the mechacon, please verify that also :)

GenuineHili, 2023-10-11:
I checked every connection with the multimeter and resoldered every solder joint again but it still doesn't work. I even measured the flex cable for continuity. I don't know what I have done wrong. Is there a chance that the xStation is faulty?

ramapcsx2, 2023-10-11:
There's always a chance that something is broken, however the xStation design is sturdy, with a very good track record.
You can assume that it works fine for now.

Can I ask you to produce a video that shows what happens in the CD player test?
If i can see what the LEDs on the xStation are doing, that would be perfect. You would need something to press down the lid switch detector for that, once in the CD player.

GenuineHili, 2023-10-12:
Ok it finally works. The solder point for pin 7 on the QSB goes in two directions. One is a via down below and the other one goes to a via that ultimativly leads to the mechacon IC. For whatever reason it wasn't making a contact to that via (maybe a broken trace?). So I manually wired it up and now it works like a charm.
Thank you very much for your support and patience, I really appreciate it.

closed 2023-10-12 by GenuineHili

ramapcsx2, 2023-10-13:
Glad to hear, enjoy! :)

I forgot what "7" connects to, but basically every signal on the QSB is important.
The bunch around "7" generally makes the ODE work at all, and the bunch around "IO4" is specifically for data to work. Just an aside :p