x-station/xstation-issues #324: Dualshock 2 doesn't work for navigating the Xstation menu

Enecororo, 2023-10-08:
At least last time I checked it doesn't.

Since Dualshock 2 works just like a regular Dualshock when plugged into a PS1, you'd think that would be covered.

Honestly wouldn't be an issue with me if my only Dualshock 1 didn't have chewed up left stick.

Is it just that one dualshock 2 in particular or is that a legit oversight?

ramapcsx2, 2023-10-08:
Hm, I don't recall that there was any issue, but I also don't have one to test right now.. I can ask around a bit :)

Enecororo, 2023-10-08:
Last time I tried to use a Dualshock 2 to play on my Xstation it didn't respond at all in the loader. that was almost a year ago though.

Hope you can get an answer

Enecororo, 2023-10-08:
I just tried plugging my Dualshock 2 into my Xstation PS1 and its working now analog on or off.

Strange. I very distinctly recall plugging in my Dualshock 2 to play Nightmare Creatures and it did nothing on the menu. Maybe I did something wrong.

Enecororo, 2023-10-08:
Closing the issue but leaving it like this in case someone can maybe figure out what might have happened

closed 2023-10-08 by Enecororo

ramapcsx2, 2023-10-09:
Oooh, right, yes, we did fix one pad problem quite some time ago, and maybe you tested then and never got the update until now :)

Enecororo, 2023-10-09:
I don't think I was on the base firmware back then and I don't recall having installed an update since, but hey my memory is really bad so maybe it was before I installed the new firmware